ashtuchkin/vive-diy-position-sensor #2: Create a more convenient method of providing base station geometry.

ashtuchkin, 2016-11-13:
Base station positions and direction matrices are hardcoded in geometry.cpp (`lightsources`). 

I'm currently using an [OpenVR](https://github.com/ValveSoftware/openvr) hello world sample on my main machine to get it (by inspecting m_rmat4DevicePose in debug mode). This is obviously not sustainable.

Subtasks:
 [ ] Keep the matrices in EEPROM.
 [ ] Create interface to set them via USB.
 [ ] Create a way to determine the matrices for a setup - special application?

paralin, 2016-12-01:
The real Vive controllers and headset do it automatically, since they have a lot of sensors and know the exact geometry of all of those sensors, it must be quite easy to derive the positions of the base stations. The room calibration is primarily used to figure out the ground + play area, I suppose. The official sensors can determine the base stations relative positions without calibration or pre-set values. I think it would be possible for us to do this as well *if* we have multiple sensors on the quad with pre-calibrated positions

mpinner, 2016-12-08:
I'd be interested to ease the process by which we're to determine the positions and direction matrices. I got the OpenVR project working am able to inspect the m_rmat4DevicePose as well. The data from there (int[16]) didn't match format of our lightsources[2] (See Below)

I was thinking i could build a little windows app that outputs the correct lighthouse calibration information (direction and position). i agreed the eeprom is a nice way to go. i'll look into something there as well.

thanks for all your work and documentation. 

[image]

ashtuchkin, 2016-12-08:
Yeah, as you can see the matrix there is 4x4 with the last column being
(0,0,0,1). For lightsource I'm using the top-left 3x3 sub-matrix (rotation)
and 1x3 position which is the last row of the 4x4 matrix above.

mpinner, 2016-12-12:
Super. this worked great. Thanks! 

all the HTTCC devices are in there. it is fascinating getting the positioning matrices for all the device.

 i wonder what it takes to build your own into the system?

ashtuchkin, 2017-01-23:
I updated the project to allow runtime configuration (with data stored in eeprom) and created a small app to easily get the base station geometry in needed format (https://github.com/ashtuchkin/vive-diy-position-sensor-geometry-getter). Will update the docs soon.

closed 2017-01-23 by ashtuchkin