ashtuchkin/vive-diy-position-sensor #4: TS3633-CM1 Module

abencomo, 2016-12-05:
Hi Alexander,

Will it possible to use the [TS3633-CM1 Module](https://www.triadsemi.com/product/ts3633-cm1/) instea of building your own [sensor board](https://cloud.githubusercontent.com/assets/627997/19845384/a4fce0e4-9ef3-11e6-95e4-6567ff374ee0.png)?

ashtuchkin, 2016-12-05:
I haven't tried it yet, but it should.

abencomo, 2016-12-06:
I ordered a few of the TS3633-CM1 Module so I'll let you know if it works.

I noticed that you use three(3) photodiodes optical sensors, but it seems that only one pin is connected from the Teensy to the TLV2462IP.  Are you fussing the values from the optical sensors?

BTW, I was able to build your project (i.e. the firmware) using eclipse as shown below.

[image]

ashtuchkin, 2016-12-06:
Cool! I think I'll be able to experiment with TS3633-CM1 as well, but I definitely want to hear from you as well!

Yeah, those 3 photodiodes are connected in parallel to make it possible to cover the whole top hemisphere. It works fine, as the beam is almost as large as the gap between them and it looks like just a wider pulse.

I'm glad you compiled it in eclipse - did you need to do anything special? Should I add additional instructions?

abencomo, 2016-12-07:
No, I didn't.  You did such a good job setting up the code that everything worked out of you repo.  BTW, are you in the bay area near SF?

ashtuchkin, 2016-12-07:
That's great :) yeah I live in SF

abencomo, 2016-12-07:
Great, I'm at Ames (Moffett Field) near you.  I should have the TS3633-CM1 by Thursday or Friday. Cheers!

abencomo, 2016-12-20:
Hi Alex,

The TS3633-CM1 modules are in backorder so they haven't arrived yet.  In the meantime, I want to confirm a few things.

Are you using only one Vive Base Station in your video?  If that is the case, then did you switch it to mode "A"?

Is the Vive Base Station connected to your laptop via Bluetoorh?

And, do you need to have SteamVR running in your laptop?

Cheers!

ashtuchkin, 2016-12-20:
Hey Al,

> Are you using only one Vive Base Station in your video? If that is the case, then did you switch it to mode "A"?
No, both are used. It's actually exactly the same setup as I use for regular Vive games. Modes B and C

> Is the Vive Base Station connected to your laptop via Bluetoorh?
I have a gaming desktop and they are probably connected via Bluetooth, yes.

> And, do you need to have SteamVR running in your laptop?
I have SteamVR running, but it's just to wake up the stations. AFAIK there's no realtime communication between the stations and desktop.

abencomo, 2016-12-20:
Hum? That means you have a HMD in the loop.  Darn it.  It would have been great if only your IR receiving sensor with the Teensy and the base stations were the only components required.

ashtuchkin, 2016-12-20:
Well, after waking stations up I can shut down desktop and hmd, so it's not
involved. Only base stations and sensor are needed.

ashtuchkin, 2016-12-20:
You can also set up base stations to not go to sleep at all

abencomo, 2016-12-20:
Keeping them from going to sleep is a good suggestion.  However, you still need the hmd to initiate the system before the quadcopter takes off, correct?

I also noticed that in your videos you're using a Mac laptop, were you able to run SteamVR on the Mac?

Thanks Alex for taking the time to clarify all these things.

ashtuchkin, 2016-12-20:
​I'm not sure, but just connecting base stations to power seems to be
enough to initiate. The system's geometry is a different ​case - for now,
you need to calibrate the room using Vive's calibration and copy/paste the
matrices to the sensor. After that, only base stations are needed.

AFAIK SteamVR is not supported on Mac. I have a Windows gaming pc to play
vive.

mpinner, 2016-12-20:
the TS3633-CM1 Module works great!

@ashtuchkin i made a blog post highlighting your work:
https://blog.crashspace.org/2016/12/position-aware-device/

thanks again!

abencomo, 2016-12-20:
That is great. Thanks @mpinner for confirming that!  Now I cannot wait until mine arrived. Santa please, I have been good this year.

ashtuchkin, 2016-12-26:
I just received TS3633-CM1 module and checked it's output with oscilloscope, looks great, should work with the code in this repo with minor adjustment due to inverted envelope:

[image]
Channel 1: TS3633-CM1
Channel 2: Valve's own sensor (Alan Yates sent me a bunch)
Channel 3: My DIY sensor.

abencomo, 2016-12-31:
Hi Alex,

Are you planning to update the repository with those minor adjustments to handle the inverted envelope?

ashtuchkin, 2017-01-01:
Yep, as soon as I'm back from a vacation in mid jan. Open for PRs if you
need it faster, though.

erwincoumans, 2017-01-07:
Very cool, can you explain how to connect the TS3633-CM1 to Teensy and what minor changes are needed?

erwincoumans, 2017-01-07:
Great, I used Adrian Wong's change and connected the TS3633-CM1 to Teensy 3.2 and it works.
See image at the bottom here: https://github.com/erwincoumans/vive-diy-position-sensor#ts3633-cm1

How can we connect multiple TS3633-CM1 sensors?

almostsquare, 2017-01-10:
Pull request for inverted envelope along with changes in anticipation of servicing 4+ TS3633-CM1 sensors is available at https://github.com/ashtuchkin/vive-diy-position-sensor/pull/10. Instructions on how to modify board wiring to run side-by-side is provide in the PR.

This is my first PR, so please provide any feedback. I may not be structuring it correctly, so I'm happy to change things around as needed.