ashtuchkin/vive-diy-position-sensor #7: Questions: Trying to reproduce the setup

imuguruza, 2016-12-15:
Hey there,

I'm trying to reproduce your results. Firstly I'm checking that the hardware I have mounted is OK. I have reproduced you schematics in a board that looks like this:

[image]

I have placed 2 photodiodes, turned ~180º a positioned one of each lighthouses in fron of each photodiode. Once turned on, I'm seeing the next signal at the output of the 2nd OpAmp:

[image]

Classifying the signals:
[image]

I have noted down sync pulses and hits. As can be observed, in the 3rd & 4th, there are two hits. I'm not sure why this happens.

This are my doubts:

- The 2 sync pulses are not separated in 400us, I'm seeing ~2.4ms:
- In 3rd and 4th group of pulse, I'm seeing 4 pulses( 2x sync + 2x hits), does this means I'm seeing a kind of reflection or that in that period I'm reading in both photodiodes the ray which has been sent by one of the lightshouse? Are you also reading something like this?

Calculation wise, I do not have clear if knowing the source of each pulse if its crutial or not. Meaning if it is important to know if the ray has been generated by lighthouse A or B and if we need to know if it's a vertical or horizontal. Is there any way of knowing this? Is it crutial?

Great job btw

ashtuchkin, 2016-12-16:
Hey, great job, it's almost working! From the pictures you posted, it looks like the stations are not in sync - the distance between pulses should be ~400us, I've never seen it different. Is the current setup working with the Vive? Do both stations have green LEDs? Do you have one of them set to "B" and the other to "C"?

> In 3rd and 4th group of pulse, I'm seeing 4 pulses( 2x sync + 2x hits), does this means I'm seeing a kind of reflection or that in that period I'm reading in both photodiodes the ray which has been sent by one of the lightshouse? Are you also reading something like this?

Probably reflection or sync problems. When it's the same sweep, the difference between pulses is on the order of 10us.

> As far as I understand, the pulses generated using 2 lighthouses are: ...

Yes, that's right.

> Calculation wise, I do not have clear if knowing the source of each pulse if its crutial or not. Meaning if it is important to know if the ray has been generated by lighthouse A or B and if we need to know if it's a vertical or horizontal. Is there any way of knowing this? Is it crutial?

Well, it's crucial to map it to geometry to get the actual position. It's easy to determine which sync pulse is coming from which station by analyzing their width, see table on [this page](https://github.com/nairol/LighthouseRedox/blob/master/docs/Light%20Emissions.md). skip is station, rotor is horizontal/vertical. data can be ignored.

imuguruza, 2016-12-16:
>From the pictures you posted, it looks like the stations are not in sync - the distance between pulses should be ~400us, I've never seen it different. Is the current setup working with the Vive? Do both stations have green LEDs? Do you have one of them set to "B" and the other to "C"?

You're right, I had the stations ins "A" and "B". I have switched to "B" abd "C" and now I'm seeing the pulses right. The sync pulses with a delay of 400us and a sinlge hit per sweep

imuguruza, 2016-12-16:
>Well, it's crucial to map it to geometry to get the actual position. It's easy to determine which sync pulse is coming from which station by analyzing their width, see table on this page. skip is station, rotor is horizontal/vertical. data can be ignored.

So if I get it, it means that:

- j0 & j1 -> Active lighthouse & Vertical sweep.
- k0 & k1 -> Active lighthouse & Horizontal sweep.
- j2 & j3 -> Inactive lighthouse & Vertical sweep.
- k2 & k3 -> Inactive lighthouse & Horizontal sweep.

So I should never see, in the same sweep, two sync pulses that belongs to j0/1 or k0/k1.
I have observed that "inactive" sync width signals happens for both sync signals and that photosensors also reads ray. Should we ignore this event??

ashtuchkin, 2016-12-16:
It should be
'''
"active" sync station B -> "inactive" sync station C -> horizontal sweep
pulse station B
.. same .. vertical pulse
"Inactive" sync station B -> "active" sync station C -> horizontal sweep
pulse station C
.. same.. vertical
'''

Also keep in mind that pulse lengths in that table are not absolute -
because of the filters and non-ideal opamps it can differ, but all in the
same amount. I have some code to adjust it in the repo.

imuguruza, 2016-12-19:
Yeah, now I'm able to see pulses correctly. Thanks for you help!

closed 2016-12-19 by imuguruza