ashtuchkin/vive-diy-position-sensor #8: Time offset doubt

imuguruza, 2016-12-20:
Hey there,
I continue understanding how the calculations works.
Just a detail that I'm not sure:
Following the [calculation page](https://github.com/ashtuchkin/vive-diy-position-sensor/wiki/Position-calculation-in-detail), I have a doubt about the timing called `< time offset in us>`.

Is supposed the timing since the sync pulse is generated, or is ended?
I think graphically explained is more clear, this value is `x`, `y`or `z` delay (in us)?

[image]

I don't see it in the sourcecode...

Thanks!

ashtuchkin, 2016-12-20:
Great question! It's Z. We start the "cycle" when the first sync pulse
starts.

Note, that this is independent on which station is currently
laser-sweeping, the timing is always counted from the first sync pulse,
which can be produced by a different station.

imuguruza, 2016-12-21:
OK, thanks for the info

closed 2016-12-21 by imuguruza