ashtuchkin/vive-diy-position-sensor #9: Explanation

Bloodline95, 2017-01-07:
Hi,
I'm currently trying to get it working on a XMC 4500 Relax Kit, because I have to use an Infineon controller for my Bachelor Thesis. My current problem is, that I always get a negative angle between -1.2 and -1.3°.
So could you please explain the formular for the angle. Right now I don't really get all of it completely.

ashtuchkin, 2017-01-07:
Hey, which particular angle are you referring to? Have you seen this:
https://github.com/ashtuchkin/vive-diy-position-sensor/wiki/Position-calculation-in-detail

Bloodline95, 2017-01-07:
Yeah, I mean the first formular. I get my measured time in ns, so I divide it by 1000. But I always get a negative angle between -1.2 and -1.3°.

ashtuchkin, 2017-01-07:
Negative angle in general is fine; it just means that the sensor is to the left of center line (if it's a horizontal sweep).

What is the range of timings you get? Maybe the problem is in the timing module or the unit conversion? 

The formula itself should be quite simple: `angle = (time_in_microsec - 4000) * PI / 8333.3`

Bloodline95, 2017-01-10:
Here is a OneDrive link to the data I get. I fear that the code I use to get the time intervals is wrong.

[https://1drv.ms/u/s!At2vyt_CVX2WlwpsfbrrJKduDgek]()

`//calculate lenght of the sync pulse`

`i[0] = last_signal_period;`
`i[1] = last_signal_duty;`
`sync_pulse = (i[0] - (i[1] / 100 * i[0])) / 1000;`
`//sync pulse decoding`
`if(decode())`
`{`
`	angle =(i[0] / 1000 - 4000) * M_PI / 8333;`
` }`

I'll write a new code today with another methode to get the timings.

ashtuchkin, 2017-01-10:
Yeah the timings seem wrong. Too much variation. Be sure to calculate timing from the beginning of sync pulse to the center of the laser sweep cycle.

One more thing - you're getting angle in radians, not degrees.

Bloodline95, 2017-01-10:
Could it be, that it just won't work with only one Lighthouse Box?

Bloodline95, 2017-01-12:
I got my hands on an oscilloscope. I get some really weird pulses out of my sensors. I think, that my Lighthouse Box searchs for the second one, which is simply not there.

[image]

ashtuchkin, 2017-01-29:
Did you manage to make it work? I'm wondering what the issue was.

Bloodline95, 2017-02-20:
No I'm still struggling. Is it possible that it won't work without a second box?

ashtuchkin, 2017-02-20:
Yeah, it won't work with just one box. Two needed for a 3d fix. Are you trying to get it running without a full HTC Vive set? It might be possible to get the angles with two stations, but if you want to get xyz position, then you'll somehow need to get their position and rotation very precisely, which I don't know how to do without the full set.

ashtuchkin, 2017-02-20:
Just saw you asked this question a month ago, which I missed, sorry for that!

abencomo, 2017-02-20:
Hi Alex, I just saw your first Alpha release.  Congratulations!  I'll give it a try later this week.

The release comment says, "_**(almost) everything is configurable and users can just upload the .hex file**_"

Is that because we still need to run SteamVR, calibrate the room using Vive's calibration, and copy/paste the matrices to the sensor?

ashtuchkin, 2017-02-21:
Thanks All! This is the wrong thread for this conversation, though :) don't
be afraid to create a new issue to discuss it.

As for your question - yes, we need base station geometry from steamvr, but
also how many sensors, objects, what format, where to stream, etc. See
small example in src/settings.cpp.

jguanz, 2017-02-27:
Hello,

@Bloodline95 Did you determine what was causing your unexpected oscilloscope readings?  I'm also getting something similar.  I'm using one lighthouse (no sync cable), in mode A, and here are my readings:

[image]

jguanz, 2017-02-27:
Well shoot, the sensor was too close to the lighthouse.  We pulled it back to ~3 feet and that fixed it.

closed 2021-07-06 by ashtuchkin