ashtuchkin/vive-diy-position-sensor #14: Teensy serial communication explanation

kajep09, 2017-02-23:
Hi,

I have a Teensy 3.6 and have compiled the project (on Windows) and uploaded to the Teensy (Using Teensyduino after generating the .hex file with Ninja). However, after programming and rebooting the Teensy I was under the impression I should be able to connect to it via serial (even if it doesn't output any meaningful data), but I cannot see the device as a serial port on neither Ubuntu or Windows. On Windows it shows up as "Unknown USB device (code 43)" and on Ubuntu I cannot see anything related under /dev/tty.

If I program the Teensy with a standard Arduino serial example I can connect to it over serial on both Ubuntu and Windows. 
So before I go mad troubleshooting anything related to Teensy and serial communication, I was wondering if there are some settings that I should have configured in the project before it works?

I realize that I haven't got the correct matrices for the base station poses, but I just want to verify that the program is actually running correctly on the Teensy before I hook up my sensor board and spend time on the computer with the Vive connected (at a separate location).

TL;DR. I cannot connect to the Teensy over serial after building and uploading the project. Are there any settings that need to be configured for this to work?

kajep09, 2017-02-23:
Now I tried using "TyQt" on Windows and when I try to upload the firmware it tells me that the firmware is only compatiable with Teensy 3.1 and 3.2 - not sure why I didn't think about board version compatibility before. I never got any warning or error message when uploading with TeensyDuino, so I guess I never thought about it.

Can anyone confirm that the project is only compatible with the old versions of Teensy (3.1 and 3.2)? I should probably open a new issue to discuss board version compatibility if this is the problem. Do you have any idea how much would need to be changed for it to work on another Teensy version?

kajep09, 2017-02-23:
Hi again,

Sorry for spamming so much, but I think I have mostly solved my own problem. This is the first time I've used the Teensy MCUs, but I have now changed the following lines in the teensy-arm.toolchain.cmake file to match up with the Teensy 3.6 instead of the 3.2. I don't know if the Teensyduino verison is used for anything, but I changed it to match mine (1.35) anyway.
 Now I can build and upload the firmware to my Teensy 3.6 and it shows up as a serial device. 
Through TyQt's serial monitor I can see _"Error: No inputs configured. Please enter '!' to go to configuration mode and setup at least one input."_ and _"Unknown command 'c0'. Valid commands: view, <module> <settings>, reset, reload, write, continue."_ etc.
So I guess now it's a matter of figuring out what settings to configure. 

set(TEENSYDUINO_VERSION "135" CACHE STRING "Version of the Teensyduino SDK")
set(TEENSY_MODEL "MK66FX1M0" CACHE STRING "Model of the Teensy MCU")

set(TEENSY_FREQUENCY "180" CACHE STRING "Frequency of the Teensy MCU (Mhz)")
set_property(CACHE TEENSY_FREQUENCY PROPERTY STRINGS 180 96 72 48 24 16 8 4 2)

closed 2017-02-23 by kajep09

ashtuchkin, 2017-02-23:
Yep this is exactly the way I wanted to suggest. Glad you figured it out!

Btw I'll merge the "rewrite" branch very soon so you might wanna start
using that right away. I'll also provide compiled versions of the library
as hex files. To make it easier for to use.