ashtuchkin/vive-diy-position-sensor #15: Consider specifying a license

albrow, 2017-03-07:
Firstly, thank you so much for publishing this!

Unless I'm missing something, this repo currently does not specify an open source license. I was wondering if you would consider adding one? I would suggest the [MIT License](https://opensource.org/licenses/MIT), which is what I use for most of my own open source projects, but there are a lot of other good licenses out there.

Adding a license removes any ambiguity about who is allowed to use the code, circuit diagrams, etc. and what they are allowed to use it for. It also tends to make people more willing to contribute back :)

ashtuchkin, 2017-03-08:
Hey, I'm glad it's useful. The license is mentioned in the list of features in the beginning of readme.. Do you think I need to add it somewhere else? What's the first place you look into when you want to know which license is used?

ashtuchkin, 2017-03-08:
Ok, added the file LICENSE.txt, this should be more explicit. Thanks!

closed 2017-03-08 by ashtuchkin

albrow, 2017-03-08:
@ashtuchkin thanks for adding this! Makes it a lot more explicit and easier to find.