ashtuchkin/vive-diy-position-sensor #16: Is this as accurate as vive tracker puck?

thedonquixotic, 2017-03-12:
This seems cool. I'd like to able to create some small trackers to keep track of physical objects in VR. Things like knowing where my keyboard and mouse are. But these are only accurate to 2mm? That doesn't seem very accurate. Is there anyway to improve that?

ashtuchkin, 2017-03-12:
What kind of accuracy would be enough for your task? The position can be
averaged across several samples to get better values, at the expense of
slightly sluggish response. But it's hard to get better than 0.5-1.0 mm.

As for the title of the issue - we don't know the accuracy of vive tracker
(afaik it's not out yet), so I cannot compare. I think it'll be similar.

thedonquixotic, 2017-03-12:
Aren't the controllers and headset supposed to be submillimeter tracking?
Maybe that's just marketing hype I heard though.

ashtuchkin, 2017-03-12:
> jSlFuuH68h09H2nhAJarks5rkz28gaJpZM4MaY0P

Well, technically 0.99mm is submillimeter tracking :)

Btw, it's hard to define what we actually mean by accuracy. Is it the STDEV
of position readings in a static position? Is it max difference between
actual position (measured by ruler) and position provided by sensor across
the whole tracking volume? Or difference between what HTC Vive original
tracking devices give and what the sensor gives?

These three will give quite different results.

thedonquixotic, 2017-03-17:
@ashtuchkin That makes sense. Would you say on an experiential level it is similar in accuracy to a Vive controller or headset?

I'm going to be build it anyway but I'm curious on your thoughts. I decided if nothing else I want to build 2, one for my headphones and one for the hat rack I keep my vive and controllers on. That way I can mount and dismount in VR more easily :D

ashtuchkin, 2017-03-20:
It's similar, as long as the base stations are not occluded. Currently the main downside of this diy project is it doesn't handle situation where only 1 base station is visible, unlike the vive puck. I'm doing some experiments on it, but not sure how close I can do.

Btw, I ported this project to Particle Photon board as well, so now it can be used without wires.

closed 2021-07-06 by ashtuchkin