ashtuchkin/vive-diy-position-sensor #17: Tracking with a single station

paralin, 2017-03-20:
The Vive Puck is able to track with just a single base. I'm interested in replicating this with the DIY solution. What would be required to achieve this? I'm assuming multiple sensors positioned similarly to the puck with some kind of extremely low tolerance calibration in place to know their relative positions, correct?

mwturvey, 2017-03-20:
To be clear, are you wanting to develop your own tracked object that is accessible to be games using the Vive, or are you wanting to build your own tracked object that only uses the lighthouses for a "roomscale gps" kind of solution?

abencomo, 2017-03-20:
@mwturvey  In my case, it's the latter.  However, I have 2 lighthouses. Just to be clear.

paralin, 2017-03-20:
Latter, for robot localization without communication. I think it's the most interesting part of the lighthouse concept. I'd prefer to use one light house rather than two, to avoid having to calibrate anything but the relative pose of the sensors. What do you think?

mwturvey, 2017-03-20:
Definitely possible, but it's been surprisingly difficult to get the algorithms to work well for single lighthouse tracking.  I've been working on a similar project at GitHub.com/cnlohr/libsurvive. That project's goal is to put together an entire software stack so you can use your Vive on a less capable system.  My interest is to get a high precision indoor positioning that I can use to build a self driving wheelchair.  A big part of the effort so far has been getting the pose detection using the lighthouses.  Also, work has been done to successfully reverse engineer the protocol used by the controllers and hmd. I'm not aware of any other project that is working to determine pose from one lighthouse (other than the native Valve/HTC software), but would be interested to hear of someone had heard of any other similar effort.

paralin, 2017-03-20:
@mwturvey very cool. Ideally I'd use one base station to track a MAV as it does precision landing. The pose relative to the lighthouse is all that's required really. I think with the simplification of using a single base and not dealing with trying to do room calibration or HMD rendering could allow the code to be drastically more concise. Thoughts?

mwturvey, 2017-03-20:
Adding a second base station will only roughly double the computational work.  The two reasons I'm aware of to use two lighthouses are 1) redundancy in case of occlusion and 2) the uncertainty in position is not uniform in all axes.  Specifically, there is a lot more uncertainty in the distance from the lighthouse than side to side.  (On the order of a couple centimeters vs less than a mm).  By having two lighthouses, you can get a much more precise fix.  I expect that you'll need a microcontroller that natively supports floating point calculations in order to get high frequency position info (I.e. raspberry pi) although a lesser system is likely going to be possible, too.

ashtuchkin, 2017-03-20:
Multiple sensors on rigid frame + precise calibration + single-base station algorithm would be a lot harder to implement than the current way of using one sensor, two base stations and offload calibration to htc vive. I also think you'll need accelerometers for single-base positioning, otherwise location uncertainty would be too high in the direction of that base station, like @mwturvey said.

@mwturvey nice project! What hardware do you use?

paralin, 2017-03-20:
I plan to use a filter anyway to merge the IMU data with the Vive data, taking into account the uncertainty in the Vive positioning and potentially also taking a range finder as another input. 

The issue is I would really like to reduce the amount of hardware involved. Dropping to one base helps a lot. I think the loss of precision you describe is tolerable, +/-5cm should be fine.

mwturvey, 2017-03-20:
From the numbers I've seen,  precision in the depth dimension with a single lighthouse when using the Vive software stack ( which certainly uses the imu data) is about 2cm.  
For hardware, right now I'm just using a desktop and laptop that are a few years old -- nothing fancy.  I have a separate project where I was using a an Arduino Due connected to a rigid 6 sensor array ( using the Triad part) and was able to get good angular readings there, but I'm now convinced that the sure isn't going to be well suited to crunching the floating point numbers needed for pose estimation using a single lighthouse.  The bulk of the libsurvive work right now is pretty much all on X86 he to get the current code working and stable, although there is interest and desire by some to go get it working on a micro-- especially focusing on an esp32.  As for tracking hw, I've been using a pair of lighthouses, the custom Arduino solution above, a single Vive controller, and I've got a tracker I've been trying to figure out the USB protocol for, but it appears to be more different than the controller's protocol than I expected and is slowing me down.  Pretty much everyone else wiring on libsurvive just has a standard HTC Vive setup.

closed 2021-07-06 by ashtuchkin