ashtuchkin/vive-diy-position-sensor #21: Compiler

DarioRQC, 2017-04-06:
Hi,

I'm quite new in Github and I don't understand how can I compile or modify your code.
I don't know how I have to use the Cmake 3.5+

If I use Arduino IDE compiler or Visual micro , could I program the teensy board?

Thanks!

ashtuchkin, 2017-04-06:
What OS are you using? Arduino IDE is, unfortunately, not supported, as
well as Visual Studio. Current way to compile is using command line.

DarioRQC, 2017-04-07:
Hi Alexander,

Thank you for answering.

I'm using Windows 10. How can I learn to use this programs?

What does "%path%" means?

Hope you can help me.

Regards,

Darío

DarioRQC, 2017-04-07:
How can I solve ""Please provide PLATFORM env variable"?

ashtuchkin, 2017-04-07:
Oh! Good you've got to this. I need to update the docs, but for now you
need to add environment variable named "PLATFORM" with value "teensy"
before running CMake.

For example, type these commands:
```
set PLATFORM=teensy
CMake.exe  ... the command line you provided before
```

More info on environment variables:
https://superuser.com/questions/284342/what-are-path-and-other-environment-variables-and-how-can-i-set-or-use-them

ashtuchkin, 2017-04-10:
> Please, could you share with me a screenshoot of your Cmake config?
>
> I attach my config.
>
> [image: cmake]
> <https://cloud.githubusercontent.com/assets/20417021/24853259/5d0b6ab8-1dd2-11e7-920c-7fda204c0bfb.png>
>
> I have this error:
>
> The C compiler identification is GNU 6.3.1
> The CXX compiler identification is GNU 6.3.1
> Check for working C compiler: D:/Programas/GNU Tools ARM Embedded/6
> 2017-q1-update/bin/arm-none-eabi-gcc.exe
> CMake Error at D:/Users/Dario/Dropbox/Documents/GitHub/vive-diy-
> position-sensor2/platform-teensy/teensy-arm.toolchain.cmake:24 (message):
> GNU ARM Embedded Toolchain not found. Please install it and provide its
> /bin directory as a TOOLCHAIN_BIN_PATH variable.
> Call Stack (most recent call first):
> D:/Users/Dario/Dropbox/Documents/GitHub/vive-diy-position-sensor2/bin/
> CMakeFiles/3.8.0-rc4/CMakeSystem.cmake:6 (include)
> D:/Users/Dario/Dropbox/Documents/GitHub/vive-diy-position-sensor2/bin/
> CMakeFiles/CMakeTmp/CMakeLists.txt:2 (project)
>
> CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
> CMake Error at D:/Programas/CMake/share/cmake-3.8/Modules/CMakeTestCCompiler.cmake:37
> (try_compile):
> Failed to configure test project build system.
> Call Stack (most recent call first):
> CMakeLists.txt:5 (project)
>
> is it possible to export your code to Redbear duo?
>
> Thank you,
>
> Cheers,
>
> Darío

Windows is not my main platform, but I'll try to send you cmake settings
when I return home. From the error it seems something's wrong with finding
toolchain and C compiler.

Redbear is possible, but not easy - it's a different platform from teensy
and particle, so input code would need to be reimplemented, as it's
chip-specific.

Look into platform-teensy and platform-particle for a sense of how much
code will need to written to support new platform.

DarioRQC, 2017-04-10:
Hi,

Finally, I succeed with the compiling process.

I attach you a .docx with the steps for Windows.

How can I apply the .hex to teensy?

To obtain the raw data do I need particle?

thanks,

regards,

Darío

<https://cloud.githubusercontent.com/assets/20417021/24853259/5d0b6ab8-1dd2-11e7-920c-7fda204c0bfb.png>

ashtuchkin, 2017-04-11:
> ashtuchkin/vive-diy-position-sensor/issues/21#issuecomment-293016240>, or

Happy to hear that! Teensy has an updated program that you can download
from their website: https://www.pjrc.com/teensy/loader.html

Particle Photon is just a different platform with the benefit of included
wifi module. The functionality is otherwise similar to Teensy.

DarioRQC, 2017-05-09:
Hi ashtuchkin,

I'm trying to upload the .hex to Teensy 3.2 and the loader says "too large!"

I have Teensy 3.2 like you.

May I change some parameter in Cmake?

Regards,

ashtuchkin, 2017-05-09:
Is it an alpha .hex file, or you're compiling yourself? Did you add any
code?

DarioRQC, 2017-05-10:
I have compiled your code and I have tried with a new Teensy 3.2 board and now I can upload the .hex file.

Do I have to send any command through serial?
When I turn it on with all the connexions done, nothing happens in the serial monitor.

Can you share your .hex file?

Thank you for all :)

Regards

ashtuchkin, 2017-05-10:
Yes, you need to configure it through Serial before it starts sending you
data. See
https://github.com/ashtuchkin/vive-diy-position-sensor/blob/docs/docs/manual.md
for more info.

I uploaded a previous version of "my" .hex file here:
https://github.com/ashtuchkin/vive-diy-position-sensor/releases/tag/v1.0-alpha2
  It's almost the same as the current master.

Alex.

ashtuchkin, 2017-05-21:
Closing for now. Let me know if you have more questions.

closed 2017-05-21 by ashtuchkin