ashtuchkin/vive-diy-position-sensor #22: No position data

yancysang, 2017-04-19:
I have program the code and flashed the teensy.
Input the followings:
reset
sensor0 pin 12 positive
base0 origin -1.312959 1.813745 -1.120400 matrix -0.615478 -0.028460 -0.787640 -0.006982 0.999506 -0.030659 0.788123 -0.013371 -0.615372
base1 origin 1.332837 1.999052 1.053622 matrix 0.632380 -0.055890 0.772639 0.014321 0.998067 0.060475 -0.774526 -0.027179 0.631958
object0 sensor0

stream0 position object0 > usb_seria

continue

then,in debug model i type into the followings:

geom0 count
geom0 show

however,i can only receive 
debug> geom0 show ObjectPosition0: accumulating.. debug> sensor0 pulses show Pulse0: accumulating.. ObjectPosition0: 38001ms: fix 0, pos 0.0000 0.0000 0.0000, dist 0.0000 | 39001ms: fix 0, pos 0.0000 0.0000 0.0000, dist 0.0000 | 40001ms: fix 0, pos 0.0000 0.0000 0.0000, dist 0.0000 | 41001ms: fix 0, pos 0.0000 0.0000 0.0000, dist 0.0000 | 42001ms: fix 0, pos 0.0000 0.0000 0.0000, dist 0.0000 | 43001ms: fix 0, pos 0.0000 0.0000 0.0000, dist 0.0000 | 44001ms: fix 0, pos 0.0000 0.0000 0.0000, dist 0.0000 | 45001ms: fix 0, pos 0.0000 0.0000 0.0000, dist 0.0000 | 46001ms: fix 0, pos 0.0000 0.0000 0.0000, dist 0.0000 (9 total)

How can I get the correct position information from teensy?

ashtuchkin, 2017-04-21:
What sensor do you use? If TS3633, you'll need to change polarity (positive -> negative).
If that doesn't help, look into the raw pulses data by typing `sensor0 pulses show` in debug mode.

yancysang, 2017-04-21:
I used three BPV22NF as IR Sensor.
The raw pulses to teensy is:
[image]
n config mode，it shows： 
...
OBJ0    25375    0
OBJ0    26000    0
OBJ0    27000    0
...
when I type "sensor0 pulses show" in debug mode:
Pulse0: accumulating..
and type "sensor0 pulses show" shows:
Pulse0: 0 items
My devices:
Teensy 3.2
compile based on commit 8536d80
two lighthouse

ashtuchkin, 2017-04-21:
So there's​ no pulses registered.. check pin number is correct, double
check electronics, try manually sending impulses to the pin and try to make
`sensor0 pulses show` to give you at least something.

Btw, people sometimes confuse how debug mode works. The usual flow is
following:
 1. Send command to determine what you want to monitor (I.e. sensor0 pulses
show). It will start monitoring, but won't give you outputs. There can also
be several commands to monitor several things.
 2. To get outputs once, send empty line (press enter). To get outputs
every second, send command 'c'. To get outputs more frequently, send e.g 'c
50' to output every 50ms.

ashtuchkin, 2017-05-21:
There was a bug with pulse input, which I fixed today. Please check current master and let me know if it helped. Thanks!

closed 2017-05-21 by ashtuchkin

luoxz-ai, 2022-03-12:
Can you give a breadboard wiring diagram? thanks

luoxz-ai, 2022-03-12:
[image]

luoxz-ai, 2022-03-12:
Is the light blinking because the hardware isn't connected properly?

luoxz-ai, 2022-03-12:
[image]

luoxz-ai, 2022-03-12:
Is this breadboard wiring correct?

luoxz-ai, 2022-03-12:
[image]

luoxz-ai, 2022-03-12:
The output is the same without connecting the sensor.