ashtuchkin/vive-diy-position-sensor #23: Can tracking be done on android?

GameFaceLabs, 2017-04-23:
Title says it all

ashtuchkin, 2017-04-24:
Theoretically yes. You'll need access to hardware timers, though. Precise
timing is critical.

GameFaceLabs, 2017-04-25:
Interesting. As it happens, we build a standalone VR headset powered by
Android, which we have built from scratch. I would love to ask a few
questions if you had time

>
> Theoretically yes. You'll need access to hardware timers, though. Precise
> timing is critical.
>
>
> > Title says it all
> >
> > <https://github.com/ashtuchkin/vive-diy-position-sensor/issues/23>, or
mute
https://github.com/

ashtuchkin, 2017-04-25:
> auth/AAmVHSBhM3a0PT-QWtOUdsJJ_EWcz6q9ks5ry8AsgaJpZM4NFinS

Interesting, are you planning to support HTC Vive positioning? Sure, ask
any questions you have!

closed 2017-05-21 by ashtuchkin