ashtuchkin/vive-diy-position-sensor #24: Setup Guide for Vive DIY Position Sensor

SquawkBirdies, 2017-04-23:
Hello,

I am trying to build a position sensor and I am trying to follow your instructions as closely as possible. I am using the Teensy 3.2 microcontroller and I have built the sensor board following your instructions and using identical components. I have flashed your v1.0-alpha .hex file to the Teensy. Here are my questions:

Oscilloscope measurement of sensor board output:
[image]

1. I connected the Teensy to my computer using USB and I opened a terminal to its serial port. I have tried 9600 baud and 57600 baud, but I do not receive any text.

2. If I understand you correctly, I need to use SteamVR setup to generate configuration values for my space which then have to be uploaded to the Teensy. You wrote a separate program to grab the configuration values, but once I get them, how do I upload them to the Teensy? From reading your code, there seems to be some way to do this over USB serial. 

Thanks!

ashtuchkin, 2017-04-24:
Are you using one base station? Its a bit hard to see in the oscilloscope
pic, but with two stations you should have two big pulses and then a small
one for each cycle.

USB serial is the main configuration and data interface for this project,
so be sure to set it up correctly. What OS are you using? Try to go through
a getting started guide for teensy to set up the port, if it works with
simple echo-like app, then it will work with this project as well.

Also, be sure to press enter when you connect, to get the config> prompt.

After you connect, you'll be in configuration mode. You'll​ need to
copy/paste the base station information from that other app, plus enter
which pin you use, etc. There's a doc in /docs directory to help.

unonu, 2017-04-26:
Hi ashtuchkin, I'm working with SquackBirdies.
To clarify:
When we took the picture of the scope, that was just one base station. However, our full setup uses two base stations.
USB serial has been properly set up. We presume this because we're able to communicate with the device in config and debug modes. Also, a simple Teensy sketch was able to communicate over serial just fine using the same board.
The device seems to be functioning fully, only code doesn't recognise pulses. We configured the base station geometry with data acquired from the SteamVR setup, though even with bogus geometry, I would think it would at least report *some* data.
One more thing; we ran a test sketch to see if the pin was reading any input from the amplifier at all, and it does. When we cover the sensor, the readings stop, as expected.

Any ideas?

ashtuchkin, 2017-04-27:
Hmm, probably next step for debugging this is to go to debug mode and look
whether pulses are actually identified. Can you go into debug mode and
issue smth like "sensor0 pulses show"?

More info
https://github.com/ashtuchkin/vive-diy-position-sensor/blob/docs/docs/manual.md

SquawkBirdies, 2017-04-28:
We tried using the debug command to show the pulses but it does not show anything. So, we tried installing another script to make sure that the signal was actually getting to D12. That program shows that the pulses are detectable from the Teensy. 

Thank you for your help

ashtuchkin, 2017-04-28:
So it seems InputCmp is not seeing the pulses.. are you using the alpha2
release or compile master? Alpha2 should be more stable

GameFaceLabs, 2017-04-28:
Can you recommend some lenses that can be used over the ir sensors so as to embed the ir sensors inside a device?

ashtuchkin, 2017-04-29:
No, sorry, I haven't experimented with lenses in this project.

SquawkBirdies, 2017-05-01:
I have tried both compiling the current version and using the Alpha2 release. For some reason, the program does not respond to serial commands when I use Alpha2. I can get a serial console when I flash the file that I compiled, but the sensor does not seem to detect the pulses. 

If the code was able to see the pulses but could not properly interpret them, how would it interact? Would it end up not showing anything when I type "sensor0 pulses show"?

Thank you for your help.

ashtuchkin, 2017-05-01:
Alpha2 is less flexible than master wrt line endings - can you check your terminal is sending '\n' when you press enter?

`sensor0 pulses show` should show raw pulses - at this point in pipeline they are not classified/interpreted yet. The absence of pulses means we're not registering any.

unonu, 2017-05-01:
tl;dr `pulses` shows no pulses counted, but the Teensy definitely sees edges on the pin (pin 12, verified by another sketch,) but this is all using master, so we'll retry using alpha2.

I don't have the device In front of me right now, but we tried `sensor0 pulses count` before and it always counts 0. Consequently, `show` shows nothing. This was using master. Whenever I try to use the alpha2 release, it fails to flash (I'm using tycmd, Linux 4.9, ubuntu 16.04 based distro.) Squawk was able to get it flashed before from windows, so I guess we'll retry the release instead of master. I do find it odd, however, that the `pulses` command shows nothing even if it is supposed to show raw pulses received. Whereas the basic sketch that we run does see the pulses. I double checked your comparator setup in the interest of thoroughness and it seems fine (as expected.) So I wonder where the break down is.

jrobe, 2017-05-08:
I'm having the same issue with a teensy 3.2 - we're getting a good output on the scope, and in all reality it should be getting pulses - but it just seems like the teensy isn't recognizing them. Do you have any tips? I'd be willing to go through any debugging steps that you'd provide - we're trying to get this working for a NASA competition.

jrobe, 2017-05-16:
Update - I went to the v0.1 tag, and it worked great, I'm assuming there's something wrong with the multiple board/teensy versions support in the latest version, haven't had time to look further.

ashtuchkin, 2017-05-21:
Ok, I re-checked the master branch today and indeed there was a bug with the pulse input node. I pushed a fix and checked that the current master is now working fine on my teensy. Let me know if it fixes your problem as well.

jrobe, 2017-05-21:
That fixed it! Getting good position data from the sensors now. Thanks for that - thought I was going crazy until 0.1 worked.

ashtuchkin, 2017-05-27:
Great! Happy to hear that

closed 2017-05-27 by ashtuchkin