ashtuchkin/vive-diy-position-sensor #26: I have a question about the pulse of END circle

segatecm, 2017-05-15:
I had read the doc of the pules detail. I have question about the Circle.
The End Pules length do not be wrote on it. I confused with that because it will connected with the new begin pules of THE NEXT CIRCLE.
Then there are only two pules can be seen on Oscilloscope but three.
So please tell me how to check the end pules of ONE CIRCLE.
Thanks for your great job.

ashtuchkin, 2017-05-16:
Hey,

I assume you're talking about a cycle, not a circle.

Each cycle has 3 pulses: sync pulse from station A, sync pulse from station
B and a sweeping pulse from one of them. There is no "end pulse" in a
cycle. Each cycle has fixed length - 8.333ms and they go one after another,
so there's no need for an end pulse.

Did I answer your question? If not, could you draw a picture of what you're
trying to ask?

segatecm, 2017-05-17:
Yes , It's a cycle, not a CIRCLE. My mistake.
Thanks for your reply. I understand 3 pluses now.

And i have one question, how can i confirm the sweeping pulse send by A or B?

ashtuchkin, 2017-05-17:
I have more info about it in the readme.md, but the short version is using
the length of the sync pulses - if first sync pulse is shorter than second,
then the sweeping pulse is from A, else B.

segatecm, 2017-05-18:
Thanks ashtuchkin! It's a simple way to decode Sync Pulse.

i read the [Light Emissions paper](https://github.com/nairol/LighthouseRedox/blob/master/docs/Light%20Emissions.md) from your readme.md. i still have questions about the define  of Pulse Length.

There are three states, skip, data and axis.
the axis bit equal 0 means horizontally or vertically ?
the skip bit equal 1 means the cycle must be aborted and sweeping value is invalid?

and the last question,the length(us) how to convert length(ticks)?

mwturvey, 2017-05-18:
Ticks would be the number of clock cycles given a 48Mhz clock.  The only thing special about 48Mhz is that it's the sampling frequency on the actual Vive hardware (i.e. Vive controllers, Vive tracker, and the hmd).  Also, fyi, the ticks value in that table is more precise than the time, which is an approximation.

segatecm, 2017-05-18:
Thanks mwturvey. Then can i use an arduino to sample the pulse? is it fast enough?

ashtuchkin, 2017-05-18:
Axis=0 is horizontal. 1 vertical.
Skip=0 means the same base station as giving this sync pulse will also
generate sweep pulse for this cycle. 1 means the other base station is
generating sweep pulse.

Remember, each cycle have 2 sync pulses, one from each station. Exactly one
of them will have skip=0, another - skip=1. This is how you determine which
base station is emitting the sweep pulse.

Back to your question about Arduino - it's probably possible using hardware
timers, but might be hard. Also, in my experience you need ~1 microsecond
timing precision to get 1-2 mm position accuracy.

mwturvey, 2017-05-18:
Allan Yates, who developed the lighthouse hardware made an interesting comment about using an Arduino last summer.  
https://mobile.twitter.com/vk2zay/status/746917256093806594
I'd think you could get a single sensor to read alright on an Arduino Uno using hardware interrupts, but it would require some creative coding, I imagine.  Regular digitalread calls wouldn't get you close.  
I was able to get an Arduino Due to do a pretty good job of reading 6 sensors at a time, and it could probably have scaled to twice that, easily.

segatecm, 2017-05-18:
Thanks ashtuchkin and mwturvey.I will do my receiver now!

segatecm, 2017-05-20:
Hi everyone ,i use the arduino decode the sweep value!!! and copy some values here:

Date Type:
1 - 3: Pulse length(µs)
4: sweep Pulse length - 4000 (µs)

64	126	6	-38
75	117	6	-806
64	107	6	-18
74	137	6	-826
127	80	9	-486
137	95	7	-753
64	106	6	-17
75	137	6	-825
128	72	8	-473
137	96	7	-754
85	127	6	-37
95	117	6	-803
64	127	6	-36
74	116	6	-803
106	85	8	-486
116	75	7	-732
64	106	6	-13
95	117	6	-804

i am so confused about the values. for example, i copy two lines value, and translate them to bits
64	127	6	-34  
64	106	6	-13  

bit vales here: (skip data axis)
 (0 0 0)   (1 1 0)
 (0 0 0)   (1 0 0) 

from the values, look like the sweep pulse come from same station, but why the angle values are different? Which value is right?

ashtuchkin, 2017-05-21:
Looks like you're missing some cycles here. In a working two-station system, cycles should always come in this order, using your notation of (skip data axis):
```
(0 x 0) (1 x 0)
(0 x 1) (1 x 1)
(1 x 0) (0 x 0)
(1 x 1) (0 x 1) 
.. repeat ..
```

Back to your question about different angles for the same base station - looks like your determination of start/end time is inaccurate. It could be the case if you use interrupt handlers to get time on pulse start (or end). They can be delayed by several dozen microseconds depending on other stuff happening on the microcontroller.

segatecm, 2017-05-22:
Learned from your posts, i check my data,

L1: 64	127	6	-34
L2: 64	106	6	-13
Compare the two lines.
i don't understand why the L1's second pulse larger than the L2's, but its sweeping time is smaller than the L2's ?
If the error data caused by microcontroller's delay, the L1's sweeping time must big then -13.

When the 127 translate to bits, it means (1 1 0), and the middle 1 means the 'data' bit is 1.

and the description say:
The data bits of consecutive sync pulses of a base station concatenated together yield a data structure called OOTX Frame.

And the OOTX Frame can not be used, so. i think i  this line must be droped.
L1: 64	127	6	-34

i don't know if i'm right, so please give same help!

ashtuchkin, 2017-05-22:
OOTX data is not really useful and you can ignore the data bit, but that
doesn't mean you should skip this cycle as it gives you the timing/angle
data.

In general the length of the sweep pulse shouldn't matter - it depends on
too many factors. What matters is the period of time between the center of
sweeping pulse and corresponding base station's sync pulse start. This will
give you the angle.

closed 2021-07-06 by ashtuchkin