ashtuchkin/vive-diy-position-sensor #27: No pulses detected. 50mV max signal voltage?

DonnyVerduijn, 2017-05-18:
After replicating the circuit, i found out that the BPV22NF operational amplifier does not amplify the signal towards the interrupt treshold of the teensy (3.3V). the TLV2462IP ses are 50mV on the oscilloscope. How should i proceed?

ashtuchkin, 2017-05-18:
You can probably play with the amplification factor of the second opamp by
using an adjustable resistor in the feedback line.. currently it's set to
around 16.
Otherwise it's hard to say without inspecting the schematic

DonnyVerduijn, 2017-05-19:
I can confirm that the pulses are being recognized using a simple digitalread(high) conditional test, but the pulses are still not recognized using the build?

DonnyVerduijn, 2017-05-19:
I assume only the sensor0 configuration is mandatory in order to debug the pulses with sensor0 pulses show/count? I tried to use different treshold values, but nothing works. Could it be that something else is mandatory to in order to recognize the pulses?

sensor0 pin 12 positive cmp 20

ashtuchkin, 2017-05-19:
Do you use version 1.0alpha-2 ? I have some bug reports for the current
master, maybe rolling back to alpha2 can help

Yep, only sensor0 is necessary. Double-check polarity (ts3633 is negative,
others positive)

ashtuchkin, 2017-05-21:
Note, I fixed a bug with getting the pulses in a recent commit, so you could try the master branch. Let me know if it works.

ichingxs4all, 2017-05-21:
I'm getting pulses now with the bug fix . Used ' sensor0 pulses show'  to see them.

Pulse0: 
sensor 0, time 442377561us, len 30 | 
sensor 0, time 442377596us, len 15 | 
sensor 0, time 442377621us, len 40 | 
sensor 0, time 442377669us, len 3 | 
sensor 0, time 442377677us, len 9 | 
sensor 0, time 442379290us, len 118 | 
sensor 0, time 442382085us, len 86 | 
sensor 0, time 442387627us, len 65 | 
sensor 0, time 442390417us, len 98 | 
sensor 0, time 442393766us, len 1 | 
sensor 0, time 442394199us, len 1 | 
sensor 0, time 442394222us, len 17 | 
sensor 0, time 442394249us, len 25 | 
sensor 0, time 442394288us, len 39 | 
sensor 0, time 442394329us, len 17 | 
sensor 0, time 442394366us, len 7 (54 total)

Used this config with our own Lighthouse positions:

sensor0 pin 12 positive

base0 origin -1.292450 1.728665 1.322062 matrix 0.676982 0.270789 -0.684375 -0.017758 0.935596 0.352625 0.735786 -0.226567 0.638190
base1 origin 1.457454 1.714955 -1.283873 matrix -0.777161 -0.208656 0.593703 -0.012993 0.948550 0.316359 -0.629168 0.238148 -0.739887

object0 sensor0
stream0 position object0 > usb_serial
stream1 angles > usb_serial
serial1 57600
stream2 mavlink object0 ned 110 > serial1

 But I'm not getting position coordinates nor angles

OBJ0	44412	0
ANG0	45000	0				
OBJ0	45000	0
ANG0	46000	0				
OBJ0	46000	0
ANG0	47000	0				
OBJ0	47000	0
ANG0	48000	0				
OBJ0	48000	0
ANG0	49000	0				
OBJ0	49000	0
ANG0	50000	0				
OBJ0	50000	0
ANG0	51000	0

Am I doing something wrong here ?

ashtuchkin, 2017-05-21:
Great, now we have at least some visibility into what's going on inside. It
looks like there's a lot of noise in the pulses, try increasing the
comparator threshold (sensor0 pin 12 positive cmp 40). Regular threshold is
20 and corresponds to voltage 3v3/64*value.

Another thing to try is turn off the lights and try to play with sensor
rotation, so that you maximize base station signal.

ichingxs4all, 2017-05-21:
Switched off all the lights.
I've got pretty good pulses on the scope ( see attachment). No noise visible.
When I move the sensor the pulses move nice horizontal without dropping level.

Tried several comparator levels. Best results with highest level 63.

Pulse0: 
sensor 0, time 843870837us, len 90 | 
sensor 0, time 843875213us, len 134 | 
sensor 0, time 843879171us, len 78 | 
sensor 0, time 843880830us, len 17 | 
sensor 0, time 843880879us, len 18 | 
sensor 0, time 843883545us, len 82 | 
sensor 0, time 843887503us, len 114 | 
sensor 0, time 843888772us, len 2 | 
sensor 0, time 843891879us, len 72 | 
sensor 0, time 843895653us, len 5 | 
sensor 0, time 843895838us, len 101 | 
sensor 0, time 843897497us, len 17 | 
sensor 0, time 843897545us, len 18 | 
sensor 0, time 843900212us, len 103 | 
sensor 0, time 843904170us, len 113 | 
sensor 0, time 843904286us, len 6 (22 total)

I don't get any PulseProcessor results nor Phase 

PulseProcessor: fix 0, cycle id 0, num pulses 0 0 0 1, time from last pulse 4322
PulseProcessor: fix 0, cycle id 0, num pulses 0 0 0 1, time from last pulse 4323
PulseProcessor: fix 0, cycle id 0, num pulses 0 0 0 1, time from last pulse 4323
PulseProcessor: fix 0, cycle id 0, num pulses 0 0 0 1, time from last pulse 4323

CyclePhaseClassifier: fix 0, phase 0, pulse_base_len 62.500000, history 0x0, avg error 0.0 us
CyclePhaseClassifier: fix 0, phase 0, pulse_base_len 62.500000, history 0x0, avg error 0.0 us
CyclePhaseClassifier: fix 0, phase 0, pulse_base_len 62.500000, history 0x0, avg error 0.0 us
CyclePhaseClassifier: fix 0, phase 0, pulse_base_len 62.500000, history 0x0, avg error 0.0 us

[image]

ashtuchkin, 2017-05-22:
Wild guess - are you using 2 base stations or one? One-station
configuration is not supported for now.

The pulses list look weird, I don't see needed structure there. There
should be two large pulses (len 60-170) 400us apart, then small pulse (len
< 10), then repeat. Whole cycle should be 8333us long.

ichingxs4all, 2017-05-22:
I think I found the problem.....the Lighthouses where in A en B mode. I switched it to B and C ....and now it's working !

ANG0	152925	200	0.0484	-0.2834	0.4373	-0.7485
OBJ0	152925	1000	0.9759	-0.1205	-0.3831	1.0381
ANG0	152959	200	0.0485	-0.2831	0.4374	-0.7488
OBJ0	152959	1000	0.9763	-0.1201	-0.3841	1.0379
ANG0	152992	200	0.0488	-0.2832	0.4374	-0.7485
OBJ0	152992	1000	0.9756	-0.1197	-0.3840	1.0390
ANG0	153025	200	0.0484	-0.2833	0.4373	-0.7485
OBJ0	153025	1000	0.9761	-0.1204	-0.3832	1.0377
ANG0	153059	200	0.0485	-0.2832	0.4373	-0.7488
OBJ0	153059	1000	0.9762	-0.1204	-0.3840	1.0379
ANG0	153092	200	0.0487	-0.2832	0.4373	-0.7487
OBJ0	153092	1000	0.9759	-0.1201	-0.3841	1.0385
ANG0	153125	200	0.0485	-0.2832	0.4373	-0.7485
OBJ0	153125	1000	0.9759	-0.1199	-0.3835	1.0381
ANG0	153159	200	0.0487	-0.2832	0.4373	-0.7486
OBJ0	153159	1000	0.9758	-0.1200	-0.3840	1.0385
ANG0	153192	200	0.0484	-0.2831	0.4372	-0.7487
OBJ0	153192	1000	0.9764	-0.1203	-0.3838	1.0371
ANG0	153225	200	0.0486	-0.2831	0.4371	-0.7485
OBJ0	153225	1000	0.9758	-0.1198	-0.3839	1.0379
ANG0	153259	200	0.0486	-0.2833	0.4372	-0.7486
OBJ0	153259	1000	0.9758	-0.1204	-0.3837	1.0382

DonnyVerduijn, 2017-05-22:
Is there more information available regarding the configuration of the lighthouses? I assumed that these identifiers were only available for synchronization purposes, but they seem to influence the timing of the pulses if i understand right?. Maybe a good idea to document this in the readme.

ashtuchkin, 2017-05-22:
Wow thats great
​, congrats​
:)

​T
he setup instructions for HTC Vive
​say
 that you need
​to use B and C modes.. Definitely the A mode will screw up the pulse
structure.​

​I'll mention this in the readme.​

closed 2017-05-23 by DonnyVerduijn