ashtuchkin/vive-diy-position-sensor #29: inaccurate position readings? movement along x results in difference on z coordinate.

DonnyVerduijn, 2017-06-06:
Hi, 

After some testing, we have to conclude that the position coordinates are not valid. When we move our tracked object in space along x, the position on the z dimension changes too. Is this because of inaccurate base station coordinates or do we have too look elsewhere too?

solinvictus21, 2017-09-09:
Just saw this. This is likely related to the issue that I just filed.

https://github.com/ashtuchkin/vive-diy-position-sensor/issues/34

ashtuchkin, 2017-09-10:
I'm pretty sure it's not related to #34. I'd look into base station coordinates and rotation matrices first, yes. How large is the z dimension change vs x dimension?

wouterpeetermans, 2018-01-10:
I'm having the same problem here. can you be more specific how I could troubleshoot this please?

closed 2021-07-06 by ashtuchkin