ashtuchkin/vive-diy-position-sensor #30: What happens when only one Lighthouse is in sight?

DanielArnett, 2017-07-22:
I need to get an accurate (<10cm error) position of one robot relative to another robot. I'm considering placing a Lighthouse on one robot, and this sensor on the other. I don't need depth, only X and Y. 

Would this still give position within 10 cm even with only one lighthouse unit?

ashtuchkin, 2017-07-22:
​No, currently both base stations ​must be in sight to get pulse
synchronization. This problem is technically solvable, just requiring a
rewrite of the synchronization module.

Having the above in mind, with just one base station the only thing you'll
be getting is two angles from the line along which the base station is
"looking". If depth is constant and base station is overlooking the area
where your robot is moving, then you should be able to convert it to XY
coordinates, but I'm not sure how you can do that with base station on
another robot.

Basically, if you have two robots moving in the same plane, then this won't
work at all.

DanielArnett, 2017-07-23:
I appreciate your perspective, thank you. I don't need the robots position, there are other sensors for that. I'm looking for accurate _relative_ position between two robots, say for them to dock for instance. If I can get an accurate vector from one robot to another; then this sensor would be perfect. I don't care about coordinates as much as I do the angles between robots.

Any first thoughts on where I should look if I want to disable pulse synchronization and just use the pulses from one Lighthouse?

ashtuchkin, 2017-07-23:
O
​k, I trust you know what you're doing.
​
​This is
​the piece of sync algorithm that uses both stations to classify the cycle
phase depending on the large pulse widths:

https://github.com/ashtuchkin/vive-diy-position-sensor/blob/master/src/cycle_phase_classifier.cpp#L31

Having two stations simplifies the algorithm significantly, so I chose to
do it this way. It's possible to change it to work with only one base
station, but that would require rather significant rewrite of that module.
I'd assess it would take at least a week to do that, which I don't have atm.

DanielArnett, 2017-07-23:
Okay, what about this for a workaround; how about if there were two lighthouses right next to each other; would that run and at least try to estimate position?

BoyHolthausen, 2017-07-24:
The lighthouse won't stay on if it detects its moving, so unless the robot is moving steadily at snail pace I would look for a different solution altogether. Your constraints are not too high (only azimuth and relatively low accuracy of 10cm), so you could consider making a rotating laser fan mechanism yourself.

Another reason to make your own version in this case is that the sync flashes would likely need 360 degrees coverage, but the basestations are limited to less than 180 degrees. Because you only need a single axis, you can get away with just a single flash at a constant interval however, no encoding/decoding of pulse lengths needed. Although, if multiple robots were equipped with a lighthouse-style tracking solution, you need to think about multiplexing. The easiest solution would be time division multiplexing, could use a cheap radio to communicate between robots who is allowed to emit light signals.

At this point however, it might be better to consider using a camera and OpenCV combined with some LEDs. The camera could be mounted on a servo to get 360 degree coverage, or you can use about 3 cameras with wide angle lenses.

DanielArnett, 2017-07-24:
Very interesting, thank you @BoyHolthausen! I was thinking the same thing using wide angle cameras and a visible laser. I'll think about this! Thanks

closed 2021-07-06 by ashtuchkin