ashtuchkin/vive-diy-position-sensor #33: Blinking Orange Light

DanielArnett, 2017-09-08:
I have the board assembled as shown, code built, and uploaded via the [teensyduino](https://www.pjrc.com/teensy/td_download.html) uploader but it is slowly blinking an orange light, and not outputting any serial information as expected. Any idea what it's doing, or what I could try to get it working? Really excited to experiment with this project!

ashtuchkin, 2017-09-10:
You have to configure it first, using serial connection. Refer to the manual (in works for now): https://github.com/ashtuchkin/vive-diy-position-sensor/blob/docs/docs/manual.md

SamaliLiyanage, 2018-06-25:
Hi,
I don't really understand how I could switch to the different mode i.e. Configuration mode, Debug mode etc etc

closed 2021-07-06 by ashtuchkin