ashtuchkin/vive-diy-position-sensor #34: Length of sensed sync pulses gets longer the closer you get to the Lighthouse.

solinvictus21, 2017-09-09:
I've adapted your circuit for my project, and while getting everything pulled together (months of work so far), I believe I have uncovered a slight flaw (or at least a shortcoming) in the circuit design. The closer the sensor gets to the lighthouse, the sensed timing of the sync pulses gets longer and longer than they should be. You can reproduce this behavior particularly by putting the sensor within about a foot or less, where the sensed sync pulses for even the shortest syncs (~3000 ticks) stretch beyond the range of what you should get for any of the pulses on Lighthouse station A (over ~5000 ticks).

If you have your base station mounted up high, this probably won't be a problem. My Lighthouse is mounted about waist-high and pointed toward the floor (building tiny robots that only move around on the floor) and when the robot gets within less than about 1.5-1.75 meters from Lighthouse, the problem is enough that it starts causing the software to get confused about which sync pulses are which.

My initial suspicion is that the cap on the high pass filter is too large, causing the end of the sync pulse to stretch out as more and more light from the sync pulse comes through. I'm going to mess around with switching up the RC values for the high pass filter to see if it gets better or worse, starting by using a 4.7µF cap and a 100k resistor on the high pass filter instead of a 10µF cap and 47k resistor, but figured I'd file an issue in case you have any further insight into how to address the issue.

ashtuchkin, 2017-09-10:
Hey, I can see how it can happen, yes. The software compensates for it, though. Basically I always treat sync pulse length as relative to other sync pulses, not absolute value. Within several cycles the program should be able to adjust.

All in all, I recommend using something more production-ready as a sensor, like TS3633 modules.

I'm also open to any suggestions about how to make the DIY sensor better - I'm not a pro at electrical engineering and might have messed up something there :)

solinvictus21, 2017-09-10:
Nothing for production. Just making tiny robots that need localized positioning. I looked into the TS3633 modules, but I'm unsure if I can solder the WLCSP package it comes in. Like you, I'm also not an electrical engineer and I'm relatively new at soldering. I am now able to solder down to 0603 components on pre-printed boards, which is what I use for the PCB I had printed based around your schematic (though doubled to use a quad op-amp for two diode inputs). I'm going to see if I can just get better accuracy for the sync pulses by tweaking this schematic a bit before I try another route.

solinvictus21, 2017-09-15:
Update: The problem is unrelated to the R-C ratio of the high pass filter. Switching the values as described above had no effect. However, I've confirmed that the problem is actually the TIA feeback capacitor, but I haven't decided if it's too large or too small as yet. I confirmed this by removing the feedback cap entirely and the sensed sync pulse lengths are significantly more stable, but without the feedback cap, there is too much signal noise, creating many "dead spots" at various distances and angle offsets from the center of the lighthouse. This leads me to believe that the feedback cap needs to be smaller, though it's still possible the problem is caused by overshoot due to too small of a cap. I'm going to order some 2.2pf and a 10pf caps and test each one and will post what I find as an update here.

ashtuchkin, 2017-09-15:
​Very interesting, thanks for updating this thread!​

solinvictus21, 2017-09-15:
No problem. I have another possible lead based on an examination of [a very similar schematic](https://trmm.net/Lighthouse). If you look about halfway down that linked page, you'll see a lighthouse sensor schematic that looks much like yours with a notable exception. It, too, uses a TIA into a high-pass filter into a non-inverting amp. It uses a different photodiode, though I checked the datasheets of both and they are largely similar (BPW34 is actually even a bit faster), and some of the numerics are suspiciously different such as the size of the TIA feedback resistor and the amount of gain in the last stage.

But the **interesting** part is the voltage divider on the front end. That lead me to do some reading, which lead me to [this description of the use of op-amps](http://www.radio-electronics.com/info/circuits/opamp_non_inverting/op_amp_non-inverting.php). In particular, note this quote in the last paragraph entitled "Non-inverting amplifier using single supply".

"To enable the op amp to run with just one power rail, the positive and negative rails have to be simulated by operating the amplifier half way between the rail and ground, and ensuring the decoupling is sufficient in all the required areas. This is often referred to as a virtual ground technique."

See what I mean? Interesting.

solinvictus21, 2017-09-28:
Yes! I have good news. After two weeks of experimenting (and becoming Digikey's best customer), I've come up with a new schematic that maintains extremely good timings that stay within the sync pulse timing windows across a much broader range and even provide better more consistent sweep pulses. Initial testing with my own software showed the sweep pulse timings staying within about 20 ticks at 48MHz when standing perfectly still, and that remaining drift is likely software-related.

The key was indeed to use a floating ground approach as described in the article I linked above, so I had to add an additional 5 passives to the schematic to create a ground isolation at 1.65v (half the 3.3v input supplied to the op-amps) and tweaked most of the numbers of the rest of the passives as a result to fit that portion of the circuit. It also required pulling a slightly different trick to amplify the signal in the 3rd phase of the circuit, which you'll see soon because after I get a chance to see if I can tweak it even further, I'm going to clone your repository and submit a PR with the new schematic.

solinvictus21, 2017-10-05:
I have the update for you. Rather than cloning the repository and updating the schematic directly, I'm just going to provide the schematic diagram for you via imgur. If you want the original Autodesk Eagle files, just let me know.

https://imgur.com/gallery/2KqbT

This schematic is actually a dual-diode detector so that I can determine orientation along a 2D plane, but you could easily halve it and swap out the op-amp if all you need is local position. This schematic is actually something of a mashup with the schematic mentioned here:

https://trmm.net/Lighthouse

However, I've done a TON of experimenting to improve sensitivity, reduce noise, and increase responsiveness. Here are the important changes to note with the linked schematic I'm providing:

1) This schematic uses a voltage divider at half of VCC to provide an A/C endpoint for the diode to both pull from and drain to, as recommended by the article linked above. This provided better responsiveness at distances closer to the lighthouse, but experimentation lead me to find further changes described below.

2) I eventually found that the BPC22NF diode was too slow, particularly in the trailing edge detection. I have swapped it out for the BPW34FA. Any version of the BPW34 will probably do, though I'd stick with versions that include the IR filter to reduce background noise.

3) I reduced the feedback capacitance in the TIA from 5pF to 2.2pF. I experimented with a range of capacitance values and found that 2.2pF was the best trade-off between responsiveness and reducing noise and ringing in the circuit.

4) I greatly increased the range of the high-pass filter via use of an 4.7nF/18k RC combo. I originally found that the size of the capacitor in this RC combo significantly affects responsiveness of the circuit when I first swapped the 10nF/47k resistor combo in your schematic with a 4.7nF/100k resistor combo. But I still found the circuit to be a little noisy at the 338Hz cutoff, so I increased the size of the filter until the noise went away. 4.7nF/27k was *almost* good enough, but is worse at farther distances, so I landed on 4.7nF/18k.

5) The final stage of amplification required larger gain, probably owing to the fact that the diode is now at half the voltage of the op-amp supply. Experimentation lead me to the gain you see in the schematic.

With these changes, the trailing and falling edges of the sensed sync pulse are now within the desired range up to within about a foot of the lighthouse, which is good enough for me. I still see some ringing in the circuit at close distances which I will likely try to optimize out in the future, but noise is greatly reduced and the timing of the sweep pulse is quite a bit more accurate.

If you have any questions, just let me know.

modulab, 2018-03-18:
Hi @solinvictus21 , nice circuit ! would you share the code for the two sensor implementation?

solinvictus21, 2018-03-18:
Certainly. I’ve since published all source code and 3 D models and everything else for my entire project. You’d be mostly interested in the source code in LighthouseSensor.cpp.

https://github.com/solinvictus21/Zippies

modulab, 2018-03-18:
@solinvictus21 wow! This is more or less the same thing that I was trying to make. Nice job man!

solinvictus21, 2018-03-19:
Excellent! Please join me on my project. It'd be great to have another set of hands working on it. It's taken me a year of iteration, mostly on the Lighthouse circuit hardware and 3D models, to get to the point I'm at now. I now have accurate positioning info from a single Lighthouse. Mostly focused on the PID models to make it do line following now. Send me a PM and I'll give you my email address and send you some more photos of a fully-assembled bot and more details on how to build one.

closed 2021-07-06 by ashtuchkin