ashtuchkin/vive-diy-position-sensor #36: Issue with instructions and firmware generated on a Mac.

MPardi, 2017-10-27:
Hi,

I'm having two issues following the instruction using Mac OS as the operating system platform.

First, the command "make vive-diy-position-sensor-upload" doesn't work, it's not even mentioned in as a rule in the make file. I was able to get firmware installed on the teensy using "tycmd upload", but this leads to the next issue.

Using the firmware generated by make, located in "build/platform-teensy/vive-diy-position-sensor.hex" I don't get any response from the serial monitor. No config menu pops up. I was able to use the .hex file included in your last release from February to some success, as that would generate a prompt and allow me to configure the firmware, but not the firmware generated by the lastest commit on GitHub. The reason I ask about the firmware generated on my end is because I need to do some reconfiguration of your code in order to get data to be sent over Serial1 of the Teensy, but any reconfiguration I do just gives me a firmware file that doesn't allow me to config.

Do you happen to have any experience with this issue?

Thanks!

ashtuchkin, 2017-11-01:
Hey Michael, sorry don't have a lot of spare time these days. 

For ..-upload target to work, make sure you have a "tyc" executable in /usr/local/bin. You can get it here - https://github.com/Koromix/tytools. "tyc" is an old name of this utility, which looks like now is called "tycmd". Rename should probably help, but I haven't tried that.
The target is defined here: https://github.com/ashtuchkin/vive-diy-position-sensor/blob/master/platform-teensy/teensy-targets.cmake#L79

The configuration issue could be happening because of different line endings that your terminal is sending when you're pressing "Enter".. That'd be my guess. 

In your email you're asking
> Also, going through your code, I noticed you mention the different Serial's found on the Teensy within your code, and the configuration command for setting up the firmware in the TY monitor mentions setting the stream output using "stream# angles > usb_serial". If I try to set it to "stream# angles > serial1" I get an error message. Is there anyway to make it output to that Serial line by default within your code, or should I implement that myself.

I actually had data output via a serial port 1, so it should be definitely supported. Have you tried the following configuration:

```
reset  # Reset all settings to clean state
sensor0 pin 12 positive  # update to whatever you have
base0 origin -1.528180 2.433750 -1.969390 matrix -0.841840 0.332160 -0.425400 -0.046900 0.740190 0.670760 0.537680 0.584630 -0.607540
base1 origin 1.718700 2.543170 0.725060 matrix 0.458350 -0.649590 0.606590 0.028970 0.693060 0.720300 -0.888300 -0.312580 0.336480
object0 sensor0
stream0 position object0 > usb_serial
stream1 angles > usb_serial
serial1 57600  # note you must initialize serial before use.
stream2 angles > serial1
```

I started writing a manual here https://github.com/ashtuchkin/vive-diy-position-sensor/blob/docs/docs/manual.md but never got to finish it.. maybe it'll help as well.

shigazy, 2017-12-25:
Also running into the same issue while setting up the initial configuration when building from the latest repo.

MPardi, 2018-03-08:
Hey, sorry I never replied, we actually got it working, but we weren't able to generate certain movements within out app we were developing. With the setting up and configuration, are you just measuring the distance between the sensors and the circuit at its initial position. We were basing it off a distance measured in centimeters, but have issues properly measuring rotational matrices in a real physical setting (obviously much easier if we were developing something in a game engine).

Anyway, wanted to thank you again for the assistance. The manual is helpful, so I may attempt to use that to perfect the configuration in for the purposes we're using the project for.

Thanks again!

closed 2021-07-06 by ashtuchkin