ashtuchkin/vive-diy-position-sensor #39: Arduino

bridger35, 2018-01-17:
Is there a way to change this to Arduino?

ashtuchkin, 2018-01-17:
Probably yes, for ARM-based Arduinos like Duo. But you'll have to create a
new "platform" directory with all the wiring specific to that controller.

bridger35, 2018-01-17:
Thanks, I will look into that

closed 2018-01-17 by bridger35

Aaryan90, 2018-06-01:
What advantage do ARM-based microcontrollers have over others (say ESP 8266, Arduino) for this specific implementation? I am looking to port this functionality into nrf52832 BLE chip.

ashtuchkin, 2018-06-03:
Nice! I'm accepting pull requests if you're willing to share the code.

Wrt advantages - none really, except maybe that arm is generally faster
than arduinos and I worried that arduinos won't be fast enough. I also had
more experience with arm, so chose it as the target.

jayaramanrp, 2019-06-04:
where we have to modify the code inorder to add IMU in this project?