ashtuchkin/vive-diy-position-sensor #40: Circuit layout

bridger35, 2018-01-17:
I'm fairly new and was wondering if anyone could give me a guide on which holes to place what wires through, preferably accompanied with pictures.

bridger35, 2018-01-17:
Maybe even a walkthrough video

ashtuchkin, 2018-01-18:
Sorry, the board pictures in the readme is all I have for now. It should be
pretty straightforward to use that, plus schematics to reproduce the board.

bridger35, 2018-01-18:
What about the wire going from i 10 to g6 is that just plain metal

bridger35, 2018-01-18:
and 13 H to 9 H is that a double resistor and should the ends of the resistors be twisted

ashtuchkin, 2018-01-18:
Yes yes and yes

One other option you have if you don't want to risk is to buy a sensor
TS3633-CM1 for like $10 and be sure it works.

modulab, 2018-01-30:
Hi @ashtuchkin. can you update the circuit diagram with the pin number from the TLV2462IP? i think you are using it reversed. thank you

ashtuchkin, 2019-12-09:
sorry don't have time for that, but willing to accept a PR

closed 2019-12-09 by ashtuchkin