ashtuchkin/vive-diy-position-sensor #44: Problem with the Particle Photon platform

Wyldix, 2018-02-09:
I've been successfully using your program for a Teensy 3.6 platform and everything works fine. I tried to build the firmware for the Particle photon platform but it doesn't compile.  The compiler shows an error at the "make all" command of the make file. As far as I know you did not provide any documentation about the Particle photon platform. Would it be possible to have some explications about the command you used to build the firmware like you did for the Teensy?

Tanks, Olivier

thomasgauthier, 2018-02-16:
I had the same problem. Seems like it has to do with GCC. I was able to successfully compile it by downgrading gcc-arm-embedded from 7-2017-q4 to 5.4.

To do it on MacOS 

```bash
brew cask uninstall --force gcc-arm-embedded
brew cask install https://raw.githubusercontent.com/caskroom/homebrew-cask/ed1b2f3a4182f8512dfe88102ed79b4deb770faa/Casks/gcc-arm-embedded.rb

mkdir build && cd build
cmake .. -DPLATFORM=Particle
make
```

Wyldix, 2018-02-16:
Thank you Thomas!

I was also having some errors when compiling the Particle firmware but i didn't try to downgrade GCC. I will try that! Thanks.

alexhgian, 2018-02-20:
I'm also have this issue with Ubuntu 17.10 on 5.4 and latest

Wyldix, 2018-03-19:
I tried again on W10 and Ubuntu 16.04 both with GCC 5.4 and it still didn't compile for the Particle platform. It works for Teensy. It's getting weird since it works on MacOS...

shwetankshrey, 2018-06-29:
Hi
Did anyone successfully compile the code for Particle Photon. If yes, could you please share how you did that? I am working with Linux and have tried it on MacOS as well. However it didn't compile for me even after downgrading. On Linux, the error I get is :
`make[3]: *** No rule to make target 'all'.  Stop.
platform-particle/CMakeFiles/sensor-firmware.dir/build.make:57: recipe for target 'platform-particle/CMakeFiles/sensor-firmware' failed
make[2]: *** [platform-particle/CMakeFiles/sensor-firmware] Error 2
CMakeFiles/Makefile2:121: recipe for target 'platform-particle/CMakeFiles/sensor-firmware.dir/all' failed
make[1]: *** [platform-particle/CMakeFiles/sensor-firmware.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2`

shwetankshrey, 2018-07-05:
@thomasgauthier @ashtuchkin Please help me out here. Thanks.