ashtuchkin/vive-diy-position-sensor #46: Start of second syncpulse is off(~17ms).

koen1993, 2018-03-13:
So i used the sensor board as shown, but wanted to process the signal by myself (via a esp32 DevKit v1).
Now when i read the time, the second syncpulse starts to late: 

> 1->0:    97ms      (97)  (SyncPulse 1)  
> 0->1:  416ms    (319) (Time between Pulse 1&2)  
> 1->0:  532ms    (116) (SyncPulse 2)  
> 0->1: 4218ms (3686) (LaserSweep)  
> 1->0: 4226ms       (8)      (...)  
> 0->1: 8332ms (4106) (Begining of next Cycle/end of this Cycle)  

in this example, the second pulse start 16 ms to late, and i get this for every cycle (always off by around 17ms), But the overalltime is always on spott (with a 1ms margin). Also processing the the pulse lenght of the two syncpulses, they are often off. 
In this example, SP1: 1,0,0 SP2: 1,1,0 (skip, data, axis). This eror happens every 4-6 cycles in addition i occasionally get pulses which are longer than 135ms(136-139).

Now coupled with the still correctly overall lenght, i am at a loss as to how to correct the SyncPulses, or where my Error lies. (If this does matter, Lighthouses Sync via cable(A, B).).

modulab, 2018-05-27:
can you post the code? are you using interrupts?

ashtuchkin, 2018-05-27:
My approach was to not assume there's exactly 400µs between the pulses and
just calculate distance from the laser pulse to the sync pulse of the same
station. That worked pretty well.

koen1993, 2018-06-20:
@modulab I checked everything thought, and also found another issue #34 there its stated that the problem lies in the sensorboard.
@ashtuchkin The Problem is rather which Station(the shorter Syncpulse), and which Axis is sending right now, but i changed it now with two simple tricks: 
The longer syncpulse is the one which signals skip, 
and by collecting 4 cycles(2 Syncs + Lasersweep), i can say first cycle for station 1 is hor and second cycle is vert (dont exactly remember which one comes first right now). 

With that i think i consider this topic as closed? Its not exactly the solution but the workaround worked perfect.

closed 2018-06-20 by koen1993