ashtuchkin/vive-diy-position-sensor #47: Sensor Power Consumption

robnewton, 2018-03-24:
Can anyone tell me the power consumption of the sensor without the teensy?

I am hoping to borrow this 360 deg IR sensor design for a system that does NOT use lighthouse base stations but instead a low speed TTL fan laser on a single axis. The sensor MCU and diodes must run on very low power and wake regularly to detect a laser sweep then sleep again.

ashtuchkin, 2018-03-26:
Sorry, I have no idea. You might want to consult datasheets of the mcu and
diodes.

robnewton, 2018-03-26:
Yep I've been doing that but I was hoping someone could put a scope on theirs to give real world numbers before I built it myself. No worries though and thanks for the cool project.

closed 2019-12-09 by ashtuchkin

rwender, 2019-12-09:
We (Triad Semiconductor), have a new sensor the TS4112 that integrates the photodiode and electronics into a single component. Regarding power, this part has three modes of operation. In "operation" or detection mode the part uses ~2.5mA based on SteamVR base station sweep rates. The part also has two sleep modes. In Sleep mode, the typical current drops to 0.5mA. In deep sleep mode, the worst case current is 0.15mA. The difference between the two sleep modes is the recovery time from sleep to "operational / detect" mode with deep sleep taking longer. We will have engineering samples of the TS4112 available in Feb '20.