ashtuchkin/vive-diy-position-sensor #50: Receiving sensor position data

HiruniNuwanthika, 2018-06-11:
Hi,
I have cloned your project, compiled and created the hex file. How can I get the sensor position details in to the command line? 
Teensy_loader_Cli is not working on my PC.
I'm working on windows 10 and I have used Teensy 3.2 same as you. 

Thank you!!!

ashtuchkin, 2018-06-11:
Hey Hiruni, does the official uploader app work for you?
https://www.pjrc.com/teensy/loader.html

Try to load some other program first, see if it works.

HiruniNuwanthika, 2018-06-12:
Hi Alexander,
Thank you for answering..

I have tried few programs with Teensy loader as well as the Teensyduino. 
Blink _fast and Blink_slow programs with Teensy loader
Simple program with a photo diode circuit using Teensyuino.

What should I do to run a program in teensy_loader_cli? 
Can you please help me?

Thank you!!!

jayaramanrp, 2019-05-27:
how did you create .hex file?

HiruniNuwanthika, 2019-05-27:
yes... It worked for me

closed 2019-12-09 by ashtuchkin

luoxz-ai, 2022-02-17:
How to use tycmd in windows ? 
How can I configure the module as you stated in https://github.com/ashtuchkin/vive-diy-position-sensor/blob/docs/docs/manual.md ?

luoxz-ai, 2022-02-17:
Can the arduino serial port be able to view the debug output information?