ashtuchkin/vive-diy-position-sensor #52: Support for HTC Lighthouses 2.0

Mofeywalker, 2018-06-18:
Hi, 

we have the new HTC Vive Pro set with the new lighthouses. I read that the new lighthouses use a new way to determine the position. Is it planned to support the new lighthouses in your software?

Best regards 
Moritz

ashtuchkin, 2018-06-18:
Hey Moritz,

I'm not actively developing this project, plus don't have these new
goodies, so not likely. I'm open for pull requests, though!

Alex

Mofeywalker, 2018-06-19:
Thanks Alex for answering that quickly!

Maybe someone has the new lighthouses and knows how to start to reverse engineer the protocol. Unfortunately we don't have the expertise and time to get this started.

Moritz

Kjos, 2018-09-03:
If I'm guessing, it is now creating rays at 45 degrees from the motor, so it can fit two rays on one motor. Perhaps it's sending data by alternating rays: 45 degrees, 45, 45, -45, 45, -45, -45. If it only needs to send data on reset, that could suffice (1s / (avg of 8ms and 16ms =12) per sweep = 83bits/s

closed 2019-12-09 by ashtuchkin

luoxz-ai, 2022-01-02:
TS4231  Support for HTC Lighthouses 2.0