ashtuchkin/vive-diy-position-sensor #53: File to run for configurations

HiruniNuwanthika, 2018-06-25:
Hi,

Can you give more details on which file to run to set the configurations mentioned in the link below?
https://github.com/ashtuchkin/vive-diy-position-sensor/blob/docs/docs/manual.md

koen1993, 2018-06-25:
Use [tytools](https://github.com/Koromix/tytools) (if i remember correctly the command should be 'tycmd monitor') with this you acces the teensy, to exit debug mode (in case you start in it) type '!' (this should put you in config mode).
than you can use the above linked commands in said config mode.

HiruniNuwanthika, 2018-06-25:
Thank you for your answer..
I have cloned the tytools project from  github. Any idea how to run the
tycmd?
I could get the Tycommander. But what I want is the Tycmd.

HiruniNuwanthika, 2018-06-25:
Thank you for your answer...

I used the command 'tycmd  monitor' and It started monitoring the teensy board. But when I type '!' it does not change and put me into config mode. How can I change configurations?

koen1993, 2018-06-26:
Usualy you start in config mode, you only need to type '!' if you are in debug mode. After starting 'tycmd monitor' press enter once, after it you should read either 'config>' or 'debug>' ... if you are in debug mode, `exit` it typing '!' if you are in config mode, well then you are where you want to be ...
Or do you have a different output when you press enter?
btw this is an example configuration i used. Thought you should your own Rotationmatrixes [(This might help you if you don't know how)](https://github.com/ashtuchkin/vive-diy-position-sensor-geometry-getter)
`
reset
sensor0 pin 12 positive
base0 origin -1.099456 2.181808 1.686393 matrix 0.775341 0.268522 -0.571613 -0.006164 0.908281 0.418314 0.631512 -0.320813 0.705883
base1 origin 1.467740 2.275151 -1.877902 matrix -0.730416 -0.392384 0.559041 -0.009979 0.824549 0.565703 -0.682930 0.407620 -0.606179
object0 sensor0
stream0 position object0 > usb_serial
stream1 angles > usb_serial
serial1 57600
stream2 mavlink object0 ned 110 > serial1
write
continue
o
`

HiruniNuwanthika, 2018-06-26:
Thank you very much for your support!!!!

HiruniNuwanthika, 2018-06-29:
I have done the configurations as given in your example. For base0 and base1 I have included base station data obtained for my room settings. But when I switch to the running mode, I get all values as zero. Then I tried 'geom0 show' in debug mode. It  also gave me
"pos 0.0000 0.0000 0.0000, dist 0.0000".  But I can not find the reason for not getting position data.  Can you please help me?

ashtuchkin, 2019-12-09:
Thanks @koen1993!

closed 2019-12-09 by ashtuchkin