ashtuchkin/vive-diy-position-sensor #54: Mismatch b/w coordinate axis mapping - real world and sensor values

maker-ja, 2018-06-29:
If we move the object in a particular axis wrt the ground and the light house, we get a change in all 3 axes, which is different in respect to the change given by the Vive controllers. As per our understanding there is a mismatch in the coordinate axes that unity uses and the one given by this repository. We could try to solve the problem by applying coordinate geometry on multiple points and find the shift and rotation in both coordinate spaces. Is there any other solution for the same? Please suggest a solution.

koen1993, 2018-07-11:
You are using the teensy? You used your own RotationMatrixes?(aka you configured the teensy) configuration and rotationmatrixes are referenced in issue #53

ashtuchkin, 2019-12-09:
I assume Unity applies it's own set of rotations/transformations, make sure you got them right. Otherwise I don't have a solution. It did work fine for me when I tried without additional rotations.

closed 2019-12-09 by ashtuchkin