ashtuchkin/vive-diy-position-sensor #56: Error compiling for a Photon board

shwetankshrey, 2018-07-11:
Hi

I am trying to compile the code for a Photon board. I have a 32 bit Ubuntu 14.04 setup with arm-none-eabi-gcc version 5.4.1 and CMake 3.2.2. I did the following :
`git clone https://github.com/ashtuchkin/vive-diy-position-sensor.git`
`cd vive-diy-position-sensor/`
`git submodule update --init`
`mkdir build`
`cd build`
`cmake .. -DPLATFORM=Particle`
`make`

The code starts building and target _sensor-core_ is built successfully. However for the next target _sensor-firmware_, I get an error as follows:
`Scanning dependencies of target sensor-firmware`
`make[3]: *** No rule to make target 'all'.  Stop.`
`make[2]: *** [platform-particle/CMakeFiles/sensor-firmware] Error 2`
`make[1]: *** [platform-particle/CMakeFiles/sensor-firmware.dir/all] Error 2`
`make: *** [all] Error 2`

I check and don't find an empty _particle-firmware_ in the libs directory. So I found the repository's Github URL mentioned in _platform-particle/CMakeLists.txt_ and manually clone it in that directory as follows:
`git clone https://github.com/ashtuchkin/spark-firmware.git particle-firmware`
`cd particle-firmware/`
`git checkout release/stable`
Note : There was a particle-firmware but it was empty so I deleted it and cloned this one again.

After running the make command again in the build directory, I get a new set of errors which I understand mostly to be issues related to not adhering to the C++14 standards. Some of the errors:
- `error: defaulted declaration 'constexpr TimeDelta& TimeDelta::operator=(const TimeDelta&) const'`
- `error: body of constexpr function 'constexpr uint32_t static_strlen(const char*)' not a return-statement`
- ` error: non-constant condition for static assertion`

I compare the _platform-particle/CMakeLists.txt_ and _platform-teensy/CMakeLists.txt_  and found that the Teensy CMake file contains a `set(CMAKE_CXX_STANDARD 14)` while there isn't any mention of a similar line in the Particle CMake file. I manually add the `set(CMAKE_CXX_STANDARD 14)` statement in _platform-particle/CMakeLists.txt_, remove the build directory and rebuild. **However, this is not making any differences and the make still fails with the same errors.**

I can go about the erroneous C++14 standard lines and make them compliant to earlier versions, but I do not think that would be a correct way to approach this error. I am not an expert on CMake or C++ and whatever I have been doing till now it just Googling and my intuition. However, there should be a proper solution to this. I went over through the issues and found that people have compiled the code for their Photon boards using Macs. I don't know how this could have happened but please help me out here. Thanks in advance.

ashtuchkin, 2018-07-11:
Have you tried it with CMake 3.5+, as described in prerequisites section
<https://github.com/ashtuchkin/vive-diy-position-sensor#installation-on-macos-linux>
?
--
Alexander Shtuchkin

shwetankshrey, 2018-07-12:
Yes, I have tried with CMake 3.5.2 as well. It shows the same errors as above for me.

ashtuchkin, 2019-12-09:
Yeah I think you're right about your configuration not having a C++ 14 standard enabled. Maybe there's a difference between arm-none-eabi-gcc version 5.4.1 and the 5_4-2016q3 I was using? I don't have too much time trying to debug this, sorry.

closed 2019-12-09 by ashtuchkin