ashtuchkin/vive-diy-position-sensor #57: Issue with the setup; Teensy doesn't work.

Aaryan90, 2018-07-29:
I am going for a wireless solution for the same setup, my setup connects a 3.7V battery on the Vin of a Teensy 3.1 and then I route same power through the OpAmps as well. The Teensy doesn't power up in this case. When I remove the OpAmp ICs from the circuit, the Teensy begins to blink. I am not sure what is wrong with the setup here! Do suggest any corrective measures I can take to rectify the same!

koen1993, 2018-08-07:
I remember the SensorBoard having 5v input, so i would say not enough power, so do the obvious, and disconnect the SesnorBoard, Teensy now working? Proly not enough input. Still not working? Proly problem with the Teensy not the setup. (btw the vin pin should be the only pin capable of supplying the needed 5v on the Teensy 3.1 afaik)

ashtuchkin, 2019-12-09:
Yeah it's hard to say what's going on.

closed 2019-12-09 by ashtuchkin