ashtuchkin/vive-diy-position-sensor #58: values given by vive-diy-position-sensor-geometry-getter

HiruniNuwanthika, 2018-08-28:
What are the details included in the rotation matrix of each base station?

As I understood,
In https://github.com/ashtuchkin/vive-diy-position-sensor-geometry-getter it gives
First three values are the base station center points.
What I cannot understand is the next 6 values given. (values next to the word matrix)

Please explain it to me... 
Thank you!

ashtuchkin, 2018-08-28:
It's the non-zero values of rotation matrix. A matrix is technically 3x3,
but the diagonal is zeros, so that leaves us with 6 numbers.
--
Alexander Shtuchkin

HiruniNuwanthika, 2018-08-29:
Thank you for your explanation. But it will be more helpful if  you can give me a precise answer.
 What is the rotation matrix? is it X,Y,Z coordinate positions?

Thank you!

ashtuchkin, 2018-08-29:
No, it's the rotation matrix - sines and cosines of corresponding angles. See https://en.wikipedia.org/wiki/Rotation_matrix#In_three_dimensions

I also see in the code that it expects 9 values there, sorry for confusion. 6 won't make it.

HiruniNuwanthika, 2018-08-30:
I understood it completely... Thank you!!!

HiruniNuwanthika, 2018-08-31:
According to my understanding, 
The values that are finally printed on the terminal is fix value, x, y, z, distance between 2 lines.
What is considered as the center point of those given values?

Thank you!!!

ashtuchkin, 2018-09-02:
Not sure I understand the question.. xyz is the center. Distance between
the lines can be used to get a sense of accuracy of the position estimate
(although not directly).

HiruniNuwanthika, 2018-09-04:
Thank you!!! I understood it...

Now I have another problem. Is there a method to track another set of three photodiodes? how should I do the configurations?
Thank you!!!

ashtuchkin, 2018-09-04:
If you want to track them as independent objects, then just add them to
your config like this:
```
<your base station config>
sensor1 pin 12 positive
object1 sensor1
stream1 position object1 > usb_serial

sensor2 pin 13 positive
object2 sensor2
stream2 position object2 > usb_serial
```

If, on the other hand, you want to track them as a single object, then that
is not supported, sorry.

ashtuchkin, 2018-09-04:
If you want to track them as independent objects, then just add them to
your config like this:

```
<your base station config>

sensor1 pin 12 positive  
object1 sensor1  
stream1 position object1 > usb_serial  

sensor2 pin 13 positive  
object2 sensor2  
stream2 position object2 > usb_serial  

..same thing for third and fourth ones  
```

If, on the other hand, you want to track them as a single object, then that is not supported, sorry.

HiruniNuwanthika, 2018-09-04:
I want to track them as independent objects. 
Thank you  very much for your answer......

HiruniNuwanthika, 2018-09-13:
I tried to configure for another set of photodiodes as you have explained in https://github.com/ashtuchkin/vive-diy-position-sensor/issues/58#issuecomment-418426095.
But the following error occurred. "Invalid threshold level. Supported value: 0-63".
How can I solve the problem? 

Thank you!!!

ashtuchkin, 2018-09-13:
This is where it's coming from
https://github.com/ashtuchkin/vive-diy-position-sensor/blob/master/src/input.cpp#L103

What's the config you're using?

jayaramanrp, 2019-06-14:
> If you want to track them as independent objects, then just add them to
> your config like this:
> 
> ```
> <your base station config>
> 
> sensor1 pin 12 positive  
> object1 sensor1  
> stream1 position object1 > usb_serial  
> 
> sensor2 pin 13 positive  
> object2 sensor2  
> stream2 position object2 > usb_serial  
> 

where to add the second independent object detail in the code? 
it would be helpful, if you share some knowledge.

jayaramanrp, 2019-06-14:
> where to add the second independent object detail in the code?
> it would be helpful, if you share some knowledge.

got the answer:

open cmd (in windows) where TYtools are located.
enter into config mode by pressing "!"
then type the folowing : 
reset
#enter the based station geometry (that you calculated) or you can find the default in settings in src/settings.cpp:17 to 24
 write
continue 

-enjoy

closed 2019-12-09 by ashtuchkin