ashtuchkin/vive-diy-position-sensor #65: How to use Photon firmware ?

orgicus, 2019-09-01:
Hi @ashtuchkin ,

I've just got started with this project and it's amazing !
So much to learn from this!

On that note, I'm a bit stuck configuring settings. 😅
I've tried simply pasting the example config and sending that to the serial port (8N1, 57600 bps):

```properties
reset  # Reset all settings to clean state
sensor0 pin 12 positive
base0 origin -1.528180 2.433750 -1.969390 matrix -0.841840 0.332160 -0.425400 -0.046900 0.740190 0.670760 0.537680 0.584630 -0.607540
base1 origin 1.718700 2.543170 0.725060 matrix 0.458350 -0.649590 0.606590 0.028970 0.693060 0.720300 -0.888300 -0.312580 0.336480
object0 sensor0
stream0 position object0 > usb_serial
stream1 angles > usb_serial
serial1 57600
stream2 mavlink object0 ned 110 > serial1
```

and this is the output I get:
```properties
Reset successful.
config> Validation error: Unknown/unimplemented input type: 0
config> Updated: base0 origin -1.528180 2.433750 -1.969390 matrix -0.841840 0.332160 -0.425400 -0.046900 0.740190 0.670760 0.537680 0.584630 -0.607540
config> Updated: base1 origin 1.718700 2.543170 0.725060 matrix 0.458350 -0.649590 0.606590 0.028970 0.693060 0.720300 -0.888300 -0.312580 0.336480
config> Updated: object0 sensor0 0.0000 0.0000 0.0000
config> Updated: stream0 position object0 > usb_serial
config> Updated: stream1 angles > usb_serial
config> Updated: serial1 57600
config> 
```

I had a look at the [photon pinout diagram](https://docs.particle.io/datasheets/wi-fi/photon-datasheet/#pinout-diagrams) and the Teensy one too. As far as I can tell from your images output is connected to pin 12 which is a digital pin. 

I've tried to using pin numbers (`44,40,43,41,42,3,5,4` -> `D7,D6,D5,D4,D3,D2,D1,D0`), but I've got the same error: `Validation error: Unknown/unimplemented input type: 0`

How can I configure photon's digital pin to receive output from the rest of the circuit ?

Thank you so much,
George

P.S. If it helps I can document the provided solution into `platform-particle/notes.md` or similar

ashtuchkin, 2019-09-01:
Hey George, Photon only supports "timer" input types, so try the following (note "tim" at the end):
```
sensor0 pin <pin number> positive tim
```
Where pin number is the pin you intend to use: D0-D7 -> pin 0-7; A0-A7 -> 10-17; RX/TX -> 18/19
Recommended pins: D1, D3, RX, TX
See https://github.com/ashtuchkin/vive-diy-position-sensor/blob/master/platform-particle/input_tim.cpp#L17

orgicus, 2019-09-01:
Hi @ashtuchkin 

Thank you very much for taking me one step closer! 
Adding "tim" worked!

Regarding pins, I am bit confused because I'm still using the [photon pinout diagram](https://docs.particle.io/datasheets/wi-fi/photon-datasheet/#pinout-diagrams):

[image]

[image]

[image]

I did try all of the pins listed above (and in input_tim.cpp):

```properties
config> Updated: sensor0 pin 0 positive tim
config> Updated: sensor0 pin 1 positive tim
config> Updated: sensor0 pin 2 positive tim
config> Updated: sensor0 pin 3 positive tim
config> Validation error: Pin 4 is not connected to any timer
config> Validation error: Pin 5 is not connected to any timer
config> Validation error: Pin 6 is not connected to any timer
config> Validation error: Pin 7 is not connected to any timer
config> Validation error: Pin 10 is not connected to any timer
config> Validation error: Pin 11 is not connected to any timer
config> Validation error: Pin 12 is not connected to any timer
config> Validation error: Pin 13 is not connected to any timer
config> Updated: sensor0 pin 14 positive tim
config> Updated: sensor0 pin 15 positive tim
config> Validation error: Pin 16 is not connected to any timer
config> Updated: sensor0 pin 17 positive tim
config> Updated: sensor0 pin 18 positive tim
config> Updated: sensor0 pin 19 positive tim
```

I have reverted back to pin 3 which as far as I can tell is labeled D1 on the Photon PCB.
Also I have placed the base stations in a super simple setup:

1. base station #1 at 1.16m to the left of the photodiodes, pointing left to right
2. base station #2 at 0.25m to the right of the photodiodes, pointing right to left
3. both stations point directly at the photodiodes with no rotation at all.

The [position calculation in detail wiki](https://github.com/ashtuchkin/vive-diy-position-sensor/wiki/Position-calculation-in-detail) is super handy btw!

My assumption is the position is meters and the orientation matrix is 3x3 column major for each axis, so for no rotation at they would be identity matrices:

```log
# x           y           z
1.0000  0.0000  0.0000
0.0000  1.0000  0.0000
0.0000  0.0000  1.0000
0.0000  0.0000  0.0000

# flattened

1.0000  0.0000  0.0000 0.0000  1.0000  0.0000 0.0000  0.0000  1.0000 0.0000  0.0000  0.0000
```

I've tried this setup:

```properties
config> # Current configuration. Copy/paste to save/restore.
reset
sensor0 pin 3 positive tim
base0 origin 0.000000 0.000000 -1.160000 matrix 1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000
base1 origin 0.000000 0.000000 -0.065000 matrix 1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000
object0 sensor0 0.0000 0.0000 0.0000
serial1 57600
stream0 position object0 > usb_serial
stream1 angles > usb_serial
config> Validation successful.
config> Write to EEPROM successful. Type 'continue' to start using it.
config> ANG0.128994.0....
OBJ0.128994.0
ANG0.129000.0....
OBJ0.129000.0
ANG0.130000.0....
OBJ0.130000.0
ANG0.131000.0....
OBJ0.131000.0
ANG0.132000.0....
OBJ0.132000.0
ANG0.133000.0....
OBJ0.133000.0
ANG0.134000.0....
OBJ0.134000.0
ANG0.135000.0....
OBJ0.135000.0
ANG0.136000.0....
OBJ0.136000.0
ANG0.137000.0....
OBJ0.137000.0
ANG0.138000.0....
OBJ0.138000.0
ANG0.139000.0....
OBJ0.139000.0
ANG0.140000.0....
OBJ0.140000.0
ANG0.141000.0....
OBJ0.141000.0
ANG0.142000.0....
OBJ0.142000.0
```

So the firmware seems to get stuck in the `kKnoFix` state.

I've tried rotating `base1` by 180 degrees like so:
```
config> Updated: base1 origin 0.000000 0.000000 -0.065000 matrix -1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 -1.000000
```

but the state stays the same (slow blue blink, can't get a fix).

Can you please point me in the right direction for double checking these:

1. How can I double check if I've got the right pin ? (I'm using pin D7 on Photon, assuming it's pin 3 in ST reference). 
2. Do I need any other setup for the base stations ? (I only have base stations and powered them on, I don't currently have access to a full Vive VR setup).
3. Are there any other gotchas I should look out with this setup ?

I totally appreciate taking your time to support the project.

Many thanks,
George

ashtuchkin, 2019-09-01:
That's a great progress, congrats :)
Answering your questions:
1. For D7 you should use pin 7 in config. See the mapping in my previous reply (i.e. D0-7 -> pin 0-7). Also make sure the polarity is correct.
2. I think using just base stations should work fine. I haven't tried it myself though. Using identity matrices should be fine too at this stage - later on you'll be able to adjust it as needed.

At this stage I think using debugging mode could be helpful. When you see the results (`ANG0.128994.0....`), press Enter to switch to debug mode. You can then e.g. enter "sensor0 pulses" to output current decoded pulses once. Then "c" to make it print once a second, or e.g. "c 500" to print every 500 ms. Then you can use "o" command to return back to output mode. 
Interesting commands in debug mode:
```
sensor0 pulses
phase show
phase off
pp angles
pp bits
pp show
pp off
```

ashtuchkin, 2019-09-01:
Also, I do recommend using pins D1 or D3 (1 or 3 in config) - they have better timers connected.

orgicus, 2019-09-01:
Hi @ashtuchkin ,

Super helpful debugging tips !
This may be a silly question but how do I double check the polarity ? 😊
I have followed your advice and switched to D1 (1 in config).

I did try the commands however I didn't get the expected output:
```
#command1: sensor0 pulses
#command2: phase show
#command3: phase off
#command4: pp angles
#command5: pp bits
#command6: pp show
#command7: pp off

#command1 result: Unknown command.
#command2 result: debug> CyclePhaseClassifier: fix 0, phase 0, pulse_base_len 62.500000, history 0x0, avg error 0.0 us
#command2 result: debug> 
#command3 result: debug> Unknown command.
#command5 result: debug> Unknown command.
#command6 result: debug> PulseProcessor: fix 0, cycle id 0, num pulses 0 0 0 0, time from last pulse 0
#command7 result: debug> debug> 
```

I've noticed`PointGeometryBuilder` has `debug_cmd` and `debug_print` methods, but I haven't yet figured out how to use them. I would like debug `fix_level` to try and track down why it never seems to get to `kStaleFix` for me.

Thanks again for your continued support: much appreciated !

ashtuchkin, 2019-09-03:
Polarity would be positive if you use my schematics, or negative if you use e.g. TS3633-CM1 Prototyping Module. Basically it's the direction of the pulses.

I'd start debugging step by step. First, make sure you have the pulses on the input pin, ideally using oscilloscope, ensure that the pulses are at least 4.5 volts. If you don't have oscilloscope, use debug mode (see below), it should show you the pulse lengths if it detects them. Until pulses are detected, all the other pipeline stages are irrelevant, as they don't get the input data.

For debug mode, I looked a bit more into the code and it works like this (sorry I wasn't accurate before):
```
# Start accumulating pulses in an internal array
sensor0 pulses show
# should output "accumulating.."
# Then, you can print accumulated values by pressing "Enter".
# You should see something like "Pulse0: <value> | <value> | ...  (<num> total)"
# You can press enter several times to get most recent values. 
# Or you can enter "c" to output the accumulated data every second, until you press Enter again.
# To stop accumulating pulses and showing them after each "Enter", use
sensor0 pulses stop

# If you're only interested in counts, not the values themselves, use
sensor0 pulses count
# it can also be stopped by using 
sensor0 pulses stop
```

You need to ensure you have a steady stream of pulses before checking other stages in the pipeline. Let me know how it goes.

orgicus, 2019-09-05:
Thank you very much for the detailed information.

I will follow up in a week or so after I do my homework and test the circuit with the oscilloscope first :)

orgicus, 2019-09-19:
@ashtuchkin I borrowed a scope and tried to have a look at what's coming in.
I'm not an electrical engineer this is the first time I use a scope, so I hope captured something useful:

[image]

It does look like data to me. I see a clear difference when I turn the base stations off.

I need to mention that I only have the base stations, I don't have any of the other parts (like the HMD). Having a look at issue #4 I didn't quite understand if I'll need it or not.

I've got a Teensy 3.2 in the meantime and flashed it with the firmware and I've got a very similar issue: can't get a fix. 

This is the configuration I've used on Teensy:
```
reset
sensor0 pin 16 positive tim
base0 origin 0.000000 0.000000 -1.160000 matrix 1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000
base1 origin 0.000000 0.000000 -0.065000 matrix 1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000
object0 sensor0 0.0000 0.0000 0.0000
serial1 57600
stream0 position object0 > usb_serial
stream1 angles > usb_serial
stream2 mavlink object0 ned 110.0 > serial1
```

The physical location are mostly the same (a few centimeters off where the sensor sits in between 2 base stations on the desk).

Still `pp show` prints `PulseProcessor: fix 0, cycle id 0, num pulses 0 0 0 0, time from last pulse 0`.

Can you please help me double check the following:

1. do I need an HMD or any other accessory ? (base stations 3x3 IR grid is on, status LED is green and I see a 7-segment display showing the number 6).
2. how does pin assignment work ? do they match with Teensy pin numbering ? (e.g. for Teensy with the circuit output connected to what the Teensy datasheet shows as pin 12 results in `Validation error: Pin 12 is not supported for 'timer' input type.`, I've also tried pins 3, 6, 9, 16 with no joy and confusion over which pin number in firmware means which pin the physical board)
3. does the firmware need to get a fix one or both stations ? if the translation data in the base is a bit wrong (e.g. z translation a bit off), will the firmware still pick up the base stations, but just output offset positions ?)

Many thanks,
George

ashtuchkin, 2019-09-25:
Okay the scope shows something resembling pulses, but I can't say for sure because I need a larger picture (the screenshot you posted is 5ms wide, but the full cycle is 8.333ms). See https://github.com/ashtuchkin/vive-diy-position-sensor#how-it-works--details for what you should expect to see in a working system.

As for configuration, I'd start with just a single stream, `stream0 angles > usb_serial`. When you see the angle outputs, you'll be able to add `position` output. Mavlink is usually not needed.

For positioning, I'd move the base stations a bit away, maybe at least 1-2 meters from the sensor, ideally looking in perpendicular directions. They are not designed to be used in small spaces. Also make sure they can "see" each other - that way they'll synchronize. Alternatively, I think you can use a sync cable between them if you have it. My base stations show "b" and "c" in my setup. Make sure you follow these instructions for setting them up https://www.vive.com/us/support/vive/category_howto/installing-the-base-stations.html

Now to your questions:
> do I need an HMD or any other accessory ? (base stations 3x3 IR grid is on, status LED is green and I see a 7-segment display showing the number 6).
I honestly don't know. I haven't checked it completely without the HMD. It does work when the HMD is powered off and disconnected, but I'm not sure whether it's needed for the setup phase.

> how does pin assignment work ? do they match with Teensy pin numbering ? (e.g. for Teensy with the circuit output connected to what the Teensy datasheet shows as pin 12 results in Validation error: [...]
They are exactly the teensy board pin numbers. The thing is, not all pins are supported in the hardware. 
I used comparator input type (configured as e.g. `sensor0 pin 3 positive cmp`) for my experiments because it allows you to tune the threshold level for the pulses.
Teensy 3.2 has 3 comparators available which can be connected to pins 3, 4, 9, 11, 12 and some others (see [here](https://github.com/ashtuchkin/vive-diy-position-sensor/blob/master/platform-teensy/input_cmp.cpp#L43) for the full list). 
Threshold level is a number between 0 and 64, that correspond linearly to the range from 0V to 3.3V. It can be set in configuration like this: `sensor0 pin 3 positive cmp 32` - in this case 32 is the threshold. In your case I see there's some noise below the 2V level, so I'd recommend setting threshold 2V/3.3V*64 = 38.

You can also use Timer (`tim`) input type for some other pins (see supported pins [here](https://github.com/ashtuchkin/vive-diy-position-sensor/blob/master/platform-teensy/input_ftm.cpp#L9)).

> does the firmware need to get a fix one or both stations ? if the translation data in the base is a bit wrong (e.g. z translation a bit off), will the firmware still pick up the base stations, but just output offset positions ?)
Firmware needs to get pulses from both stations to get a fix. Otherwise it can only know 2 angles from one station (`stream1 angles > usb_serial` will print give them).

orgicus, 2019-09-25:
I was hoping I can replicate a setup exactly based on what works,
but this allows me to learn a lot more about the whole process.

I'll go away and do my homework based on your info above. 

Thank you so much for the detailed response.

orgicus, 2019-10-15:
@ashtuchkin Woo hoo! Finally got some plausible data ! 🎉 

Using a full vive setup it was much easier to use your excellent vive-diy-position-sensor-geometry-getter utility.

I've used this setup:

```
reset
sensor0 pin 12 positive cmp 32
base0 origin 0.000000 0.344039 0.000000 matrix 0.999256 0.038564 -0.000000 -0.037983 0.984184 0.173031 0.006673 -0.172902 0.984916
base1 origin 0.394682 0.289385 -5.338341 matrix -0.999873 -0.007382 0.014130 -0.007419 0.999969 -0.002584 -0.014110 -0.002688 -0.999897
object0 sensor0 0.0000 0.0000 0.0000
serial1 57600
stream0 position object0 > usb_serial
stream1 position object0 > usb_serial
stream2 mavlink object0 ned 110.0 > serial1
```

and I now get data like this:

```
OBJ0.2205252.1000.0.4784.-0.5143.-4.3821.0.4411
OBJ0.2205285.1000.0.4770.-0.5123.-4.3848.0.4424
OBJ0.2205285.1000.0.4770.-0.5123.-4.3848.0.4424
OBJ0.2205319.1000.0.4782.-0.5125.-4.3841.0.4403
OBJ0.2205319.1000.0.4782.-0.5125.-4.3841.0.4403
OBJ0.2205352.1000.0.4787.-0.5142.-4.3816.0.4406
OBJ0.2205352.1000.0.4787.-0.5142.-4.3816.0.4406
OBJ0.2205385.1000.0.4770.-0.5122.-4.3847.0.4424
OBJ0.2205385.1000.0.4770.-0.5122.-4.3847.0.4424
OBJ0.2205419.1000.0.4782.-0.5124.-4.3839.0.4404
OBJ0.2205419.1000.0.4782.-0.5124.-4.3839.0.4404
OBJ0.2205452.1000.0.4780.-0.5137.-4.3825.0.4414
OBJ0.2205452.1000.0.4780.-0.5137.-4.3825.0.4414
OBJ0.2205485.1000.0.4770.-0.5123.-4.3846.0.4423
OBJ0.2205485.1000.0.4770.-0.5123.-4.3846.0.4423
OBJ0.2205519.1000.0.4783.-0.5129.-4.3836.0.4406
OBJ0.2205519.1000.0.4783.-0.5129.-4.3836.0.4406
OBJ0.2205552.1000.0.4780.-0.5137.-4.3825.0.4414
OBJ0.2205552.1000.0.4780.-0.5137.-4.3825.0.4414
OBJ0.3043951.1000.0.4534.-0.5150.-4.3784.0.4539
OBJ0.3043951.1000.0.4534.-0.5150.-4.3784.0.4539
OBJ0.3043984.1000.0.4533.-0.5136.-4.3801.0.4536
OBJ0.3043984.1000.0.4533.-0.5136.-4.3801.0.4536
OBJ0.3044017.1000.0.4543.-0.5156.-4.3773.0.4526
OBJ0.3044017.1000.0.4543.-0.5156.-4.3773.0.4526
OBJ0.3044051.1000.0.4534.-0.5149.-4.3784.0.4541
OBJ0.3044051.1000.0.4534.-0.5149.-4.3784.0.4541
OBJ0.3044084.1000.0.4539.-0.5143.-4.3793.0.4526
OBJ0.3044084.1000.0.4539.-0.5143.-4.3793.0.4526
OBJ0.3047417.1000.0.4533.-0.5137.-4.3802.0.4535
OBJ0.3047417.1000.0.4533.-0.5137.-4.3802.0.4535
OBJ0.3047451.1000.0.4541.-0.5139.-4.3797.0.4518
OBJ0.3047451.1000.0.4541.-0.5139.-4.3797.0.4518
OBJ0.3047484.1000.0.4541.-0.5151.-4.3783.0.4531
OBJ0.3047484.1000.0.4541.-0.5151.-4.3783.0.4531
OBJ0.3047517.1000.0.4529.-0.5132.-4.3808.0.4538
OBJ0.3047517.1000.0.4529.-0.5132.-4.3808.0.4538
OBJ0.3047551.1000.0.4546.-0.5143.-4.3792.0.4517
OBJ0.3047551.1000.0.4546.-0.5143.-4.3792.0.4517
OBJ0.3047584.1000.0.4543.-0.5147.-4.3787.0.4523
OBJ0.3047584.1000.0.4543.-0.5147.-4.3787.0.4523
OBJ0.3047617.1000.0.4532.-0.5142.-4.3797.0.4535
OBJ0.3047617.1000.0.4532.-0.5142.-4.3797.0.4535
OBJ0.3047651.1000.0.4542.-0.5139.-4.3800.0.4519
OBJ0.3047651.1000.0.4542.-0.5139.-4.3800.0.4519
OBJ0.3047684.1000.0.4532.-0.5153.-4.3782.0.4549
OBJ0.3047684.1000.0.4532.-0.5153.-4.3782.0.4549
OBJ0.3047717.1000.0.4530.-0.5141.-4.3800.0.4542
OBJ0.3047717.1000.0.4530.-0.5141.-4.3800.0.4542
[... 133 lines cut ...]
OBJ0.3049951.1000.0.4544.-0.5152.-4.3782.0.4526
OBJ0.3049984.1000.0.4532.-0.5138.-4.3802.0.4533
OBJ0.3049984.1000.0.4532.-0.5138.-4.3802.0.4533
OBJ0.3050017.1000.0.4543.-0.5143.-4.3793.0.4522
OBJ0.3050017.1000.0.4543.-0.5143.-4.3793.0.4522
OBJ0.3050051.1000.0.4535.-0.5150.-4.3786.0.4542
OBJ0.3050051.1000.0.4535.-0.5150.-4.3786.0.4542
OBJ0.3050084.1000.0.4534.-0.5141.-4.3797.0.4538
OBJ0.3050084.1000.0.4534.-0.5141.-4.3797.0.4538
OBJ0.3050117.1000.0.4543.-0.5147.-4.3788.0.4524
OBJ0.3050117.1000.0.4543.-0.5147.-4.3788.0.4524
```
and scope output looks like a steady signal:

[image]

If I understand correctly, the last 3 parts (e.g.`-0.5147 -4.3788 0.4524`) are the XYZ position, in meters. As far as I can understand angles can be calculated too, correct ?
If so:

1. what is the simplest method of streaming positions and orientations for 1 object ?
2. how is orientation/angles expressed (e.g. rx, ry in radians ?, etc.)
3. what is the position relative to (e.g. `base0`, `base1`)

Many thanks again for all your support: amazing stuff!
🙏 🙏 🙏

ashtuchkin, 2019-10-16:
Nice progress, congrats!

This is the meaning of the data values you're getting (looks like the tab characters were lost, I replaced them with spaces):
```
Example line: OBJ0 3050051 1000 0.4535 -0.5150 -4.3786 0.4542

OBJ0 - position of object #0
3050051 - timestamp in ms
1000 - fix level (see values here: https://github.com/ashtuchkin/vive-diy-position-sensor/blob/master/include/messages.h#L23)
0.4535 - X coord
-0.5150 - Y coord
-4.3786 - Z coord
0.4542 - distance between the rays from the base stations
```
The (X, Y, Z) coordinates is the position of the sensor, in the same units as were used in base station configuration (if you've used vive-diy-position-sensor-geometry-getter utility, then it's meters).

Last value, the distance between the rays, can be used as an indirect estimate of the accuracy of this setup. It uses the same units as the position coordinates. As you probably know, the sensor position is calculated by "intersecting" the virtual rays from two base stations. In reality it's very rare that they actually intersect, so instead we calculate the location where the distance between the rays is minimal. This distance is the last value here. 0.45 is pretty inaccurate, it means the distance between rays is 45cm and the position can be off up to 45 cm from the real one. You need to try to get it closer to 0.005. Maybe retry the geometry-getter utility or reposition the base stations?

Now to your questions:
> what is the simplest method of streaming positions and orientations for 1 object ?
Orientation can't be calculated by using just one sensor, you have to have several of them (afaik HTC Vive uses ~a dozen on each of their controllers). Also, I haven't got to implementing orientation calculation in this project, so you'll have to add it yourself.
> how is orientation/angles expressed (e.g. rx, ry in radians ?, etc.)
Internally we have the angles of the rays from base stations (2 for each stations, rx and ry), they are expressed in radians. You can see them if you use
```
stream1 angles object0 > usb_serial
```
> what is the position relative to (e.g. base0, base1)
It's in the same coordinate system that the base station geometry is provided. So the origin corresponds to whatever HTC Vive derives from calibration process. For the config you provided, the origin is, I think, 0.344039 meters below the first base station, because that station's origin is `0.000000 0.344039 0.000000`). As far as I remember HTC Vive uses coordinate system where Y axis goes up; X and Z correspond to (X, Y) on the floor. But that needs to be re-checked.

There's a way to automatically convert coordinate system to NED by adding `ned <angle to north>` like it's done for mavlink stream in your configuration, let me know if you want more info about it.

P.S. Currently the lines are duplicated in the output - you can avoid that by removing the line starting with `stream1` (and adjusting `stream2` to be `stream1` if you actually need mavlink on the serial output).

orgicus, 2020-03-07:
@ashtuchkin I'm moving on to Photon and I'm running into a few problems I'm not sure how to solve.

1. When passing this config: `object0 sensor0 0.0000 0.0000 0.0000` I get an error:
`Sensor number (sensor<num>) required for object 0`
2. How can I double check the firmware has the correct WiFi credentials ?
3. Is there a checklist/guide on making sure the readings sent back will be correct ?
(for example, on Teensy, if I miuse the object up and down (Y axis), the X axis also moves quite a lot. I'm sure I can't hold the board perfectly on any axis, but the data suggests X axis drifts quite a lot:

```
OBJ0.1603126.1000.1.5016.1.9592.-1.1147.0.4510
OBJ0.1603126.1000.1.5016.1.9592.-1.1147.0.4510
OBJ0.1603159.1000.1.5842.1.9842.-1.1771.0.4421
OBJ0.1603159.1000.1.5842.1.9842.-1.1771.0.4421
OBJ0.1603193.1000.1.6730.2.0087.-1.2425.0.4301
OBJ0.1603193.1000.1.6730.2.0087.-1.2425.0.4301
OBJ0.1603226.1000.1.7716.2.0350.-1.3188.0.4222
OBJ0.1603226.1000.1.7716.2.0350.-1.3188.0.4222
OBJ0.1603259.1000.1.8292.2.0522.-1.3600.0.4222
OBJ0.1603259.1000.1.8292.2.0522.-1.3600.0.4222
OBJ0.1603293.1000.1.9271.2.0747.-1.4326.0.4129
OBJ0.1603293.1000.1.9271.2.0747.-1.4326.0.4129
OBJ0.1603326.1000.2.0313.2.0982.-1.5112.0.4050
OBJ0.1603326.1000.2.0313.2.0982.-1.5112.0.4050
OBJ0.1603359.1000.2.1183.2.1160.-1.5786.0.4000
OBJ0.1603359.1000.2.1183.2.1160.-1.5786.0.4000
OBJ0.1603393.1000.2.2153.2.1353.-1.6531.0.3924
OBJ0.1603393.1000.2.2153.2.1353.-1.6531.0.3924
OBJ0.1603426.1000.2.3047.2.1523.-1.7207.0.3828
OBJ0.1603426.1000.2.3047.2.1523.-1.7207.0.3828
OBJ0.1603459.1000.2.3830.2.1670.-1.7804.0.3738
OBJ0.1603459.1000.2.3830.2.1670.-1.7804.0.3738
OBJ0.1603493.1000.2.4589.2.1817.-1.8389.0.3666
OBJ0.1603493.1000.2.4589.2.1817.-1.8389.0.3666
OBJ0.1603526.1000.2.5011.2.1892.-1.8692.0.3604
OBJ0.1603526.1000.2.5011.2.1892.-1.8692.0.3604
OBJ0.1603559.1000.2.5069.2.1888.-1.8745.0.3620
OBJ0.1603559.1000.2.5069.2.1888.-1.8745.0.3620
OBJ0.1603593.1000.2.5309.2.1944.-1.8936.0.3623
OBJ0.1603593.1000.2.5309.2.1944.-1.8936.0.3623
OBJ0.1603626.1000.2.5421.2.1969.-1.9011.0.3609
OBJ0.1603626.1000.2.5421.2.1969.-1.9011.0.3609
OBJ0.1603659.1000.2.5366.2.1943.-1.8972.0.3628
OBJ0.1603659.1000.2.5366.2.1943.-1.8972.0.3628
OBJ0.1603693.1000.2.5486.2.1985.-1.9072.0.3656
OBJ0.1603693.1000.2.5486.2.1985.-1.9072.0.3656
OBJ0.1603726.1000.2.5312.2.1968.-1.8931.0.3665
OBJ0.1603726.1000.2.5312.2.1968.-1.8931.0.3665
OBJ0.1603759.1000.2.4780.2.1871.-1.8537.0.3718
OBJ0.1603759.1000.2.4780.2.1871.-1.8537.0.3718
OBJ0.1603793.1000.2.4307.2.1809.-1.8178.0.3783
OBJ0.1603793.1000.2.4307.2.1809.-1.8178.0.3783
OBJ0.1603826.1000.2.3325.2.1653.-1.7420.0.3881
OBJ0.1603826.1000.2.3325.2.1653.-1.7420.0.3881
OBJ0.1603859.1000.2.2138.2.1443.-1.6541.0.4036
OBJ0.1603859.1000.2.2138.2.1443.-1.6541.0.4036
OBJ0.1603893.1000.2.1292.2.1269.-1.5975.0.4066
OBJ0.1603893.1000.2.1292.2.1269.-1.5975.0.4066
OBJ0.1603926.1000.2.0076.2.1035.-1.5052.0.4197
OBJ0.1603926.1000.2.0076.2.1035.-1.5052.0.4197
OBJ0.1603959.1000.1.8850.2.0770.-1.4144.0.4345
OBJ0.1603959.1000.1.8850.2.0770.-1.4144.0.4345
OBJ0.1603993.1000.1.7676.2.0490.-1.3251.0.4445
OBJ0.1603993.1000.1.7676.2.0490.-1.3251.0.4445
OBJ0.1604026.1000.1.6718.2.0242.-1.2535.0.4538
OBJ0.1604026.1000.1.6718.2.0242.-1.2535.0.4538
OBJ0.1604059.1000.1.5893.2.0010.-1.1925.0.4625
OBJ0.1604059.1000.1.5893.2.0010.-1.1925.0.4625
OBJ0.1604093.1000.1.5161.1.9778.-1.1361.0.4668
OBJ0.1604093.1000.1.5161.1.9778.-1.1361.0.4668
OBJ0.1604126.1000.1.4550.1.9575.-1.0910.0.4732
OBJ0.1604126.1000.1.4550.1.9575.-1.0910.0.4732
OBJ0.1604159.1000.1.4048.1.9395.-1.0532.0.4774
OBJ0.1604159.1000.1.4048.1.9395.-1.0532.0.4774
OBJ0.1604193.1000.1.3562.1.9200.-1.0152.0.4792
OBJ0.1604193.1000.1.3562.1.9200.-1.0152.0.4792
OBJ0.1604226.1000.1.3131.1.9020.-0.9849.0.4863
OBJ0.1604226.1000.1.3131.1.9020.-0.9849.0.4863
OBJ0.1604259.1000.1.2759.1.8840.-0.9550.0.4869
OBJ0.1604259.1000.1.2759.1.8840.-0.9550.0.4869
OBJ0.1604293.1000.1.2453.1.8683.-0.9312.0.4889
OBJ0.1604293.1000.1.2453.1.8683.-0.9312.0.4889
OBJ0.1604326.1000.1.2120.1.8495.-0.9047.0.4906
OBJ0.1604326.1000.1.2120.1.8495.-0.9047.0.4906
OBJ0.1604359.1000.1.1819.1.8313.-0.8801.0.4915
OBJ0.1604359.1000.1.1819.1.8313.-0.8801.0.4915
OBJ0.1604393.1000.1.1598.1.8172.-0.8618.0.4917
OBJ0.1604393.1000.1.1598.1.8172.-0.8618.0.4917
OBJ0.1604426.1000.1.1339.1.8010.-0.8420.0.4944
OBJ0.1604426.1000.1.1339.1.8010.-0.8420.0.4944
OBJ0.1604459.1000.1.1097.1.7855.-0.8234.0.4968
OBJ0.1604459.1000.1.1097.1.7855.-0.8234.0.4968
OBJ0.1604493.1000.1.0944.1.7751.-0.8103.0.4960
OBJ0.1604493.1000.1.0944.1.7751.-0.8103.0.4960
OBJ0.1604526.1000.1.0776.1.7652.-0.7995.0.5004
OBJ0.1604526.1000.1.0776.1.7652.-0.7995.0.5004
OBJ0.1604559.1000.1.0669.1.7579.-0.7903.0.4995
OBJ0.1604559.1000.1.0669.1.7579.-0.7903.0.4995
OBJ0.1604593.1000.1.0593.1.7535.-0.7849.0.5001
OBJ0.1604593.1000.1.0593.1.7535.-0.7849.0.5001
OBJ0.1604626.1000.1.0546.1.7509.-0.7819.0.5009
OBJ0.1604626.1000.1.0546.1.7509.-0.7819.0.5009
OBJ0.1604659.1000.1.0508.1.7487.-0.7791.0.5015
OBJ0.1604659.1000.1.0508.1.7487.-0.7791.0.5015
OBJ0.1604693.1000.1.0501.1.7487.-0.7786.0.5014
OBJ0.1604693.1000.1.0501.1.7487.-0.7786.0.5014
OBJ0.1604726.1000.1.0497.1.7498.-0.7803.0.5039
OBJ0.1604726.1000.1.0497.1.7498.-0.7803.0.5039
OBJ0.1604759.1000.1.0521.1.7515.-0.7817.0.5029
OBJ0.1604759.1000.1.0521.1.7515.-0.7817.0.5029
OBJ0.1604793.1000.1.0540.1.7531.-0.7835.0.5030
OBJ0.1604793.1000.1.0540.1.7531.-0.7835.0.5030
OBJ0.1604826.1000.1.0564.1.7549.-0.7860.0.5038
OBJ0.1604826.1000.1.0564.1.7549.-0.7860.0.5038
```
)

I can't thank you enough for your patience and guidance.
(I am keen to summize and contribute what I've learned as a guide to the repo)

closed 2021-07-06 by ashtuchkin