ashtuchkin/vive-diy-position-sensor #66: Error: reinterpret_cast from integer to pointer

toncho11, 2019-09-11:
Compilation failed on Windows using:

arm-none-eabi-g++.exe (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

Error:

FAILED: platform-teensy/CMakeFiles/vive-diy-position-sensor.dir/input_ftm.cpp.obj
D:\Work\ARM_Embedded\bin\arm-none-eabi-g++.exe  -DARDUINO=10607 -DF_CPU=96000000 -DLAYOUT_US_ENGLISH -DNEW_H -DTEENSYDUINO=127 -DUSB_SERIAL -D__MK20DX
256__ -I../include -I../libs/cores/teensy3 -O2 -Wall -ffunction-sections -fdata-sections -Wstack-usage=256 -mcpu=cortex-m4 -mthumb -mfp16-format=ieee
-fno-rtti -fsingle-precision-constant -Woverloaded-virtual   -std=gnu++14 -MD -MT platform-teensy/CMakeFiles/vive-diy-position-sensor.dir/input_ftm.cp
p.obj -MF platform-teensy\CMakeFiles\vive-diy-position-sensor.dir\input_ftm.cpp.obj.d -o platform-teensy/CMakeFiles/vive-diy-position-sensor.dir/input
_ftm.cpp.obj -c ../platform-teensy/input_ftm.cpp
../platform-teensy/input_ftm.cpp:53:6: error: reinterpret_cast from integer to pointer
     {&FTM0_SC, IRQ_FTM0, {{/*25*/NA, /*44*/22}, {/*26*/33, /*45*/23},  {/*27*/24, /*46*/ 9}, {      NA, /*49*/10},

ashtuchkin, 2019-09-11:
This gnu toolchain version didn't exist when this project was active :) I'm open to pull requests if you'd like to fix it.

toncho11, 2019-09-11:
I do not know the code... What is the GNU gcc version that works?

ashtuchkin, 2019-09-11:
5_4-2016q3, as specified in README.md

ashtuchkin, 2019-12-09:
Looks like you resolved it, closing.

closed 2019-12-09 by ashtuchkin