ashtuchkin/vive-diy-position-sensor #69: How to get the output data?

toncho11, 2019-12-05:
I tried the serial console:

> reset  # Reset all settings to clean state
> sensor0 pin 12 positive
> base0 origin -1.528180 2.433750 -1.969390 matrix -0.841840 0.332160 -0.425400 -0.046900 0.740190 0.670760 0.537680 0.584630 -0.607540
> base1 origin 1.718700 2.543170 0.725060 matrix 0.458350 -0.649590 0.606590 0.028970 0.693060 0.720300 -0.888300 -0.312580 0.336480
> object0 sensor0
> stream0 position object0 > usb_serial
> stream1 angles > usb_serial
> serial1 57600
> stream2 mavlink object0 ned 110 > serial1

But it does not stream data continuously on the serial port. I see the output from the commands but not the data streaming. 

Does it matter if the emitters are not active? I am not next to the vive stations right now, but I thought that there will always be some signal - noise sent on the serial console, right?

toncho11, 2019-12-05:
Hi,

This info is not accessible form the main page: 

https://github.com/ashtuchkin/vive-diy-position-sensor/blob/docs/docs/manual.md

toncho11, 2019-12-05:
And did "write", but still no data.

ashtuchkin, 2019-12-09:
I haven't had time to update the README, but willing to accept pull requests!

closed 2019-12-09 by ashtuchkin