ashtuchkin/vive-diy-position-sensor #73: Cannot get sync state

tech0x, 2020-09-10:
Добрый день, подскажите пожалуйста как отдебажить сенсор.
Я выставляю порог (threshold) на уровне 36, поскольку в указанной схеме ноль после усиления почему то не получается получить. И сигнал с базовых станций начинает читаться, декодер фрейма, выдает какие то циферки, которые меняются если передвигать сенсор. Но светодиод все равно мигает с интервалом в одну секунду.
По коду насколько понимаю это сигнализирует что лок станций не произошел и сама позиция не доступна.
Что можно еще ткнуть что бы понять что подправить ?

Благодарю.

ashtuchkin, 2020-09-10:
Привет! Да, медленное мигание значит нет фикса. У модуля есть дебаг-режим, см вот здесь небольшое описание: https://github.com/ashtuchkin/vive-diy-position-sensor/issues/65#issuecomment-526940376 (и ниже по треду).

tech0x, 2020-09-11:
Добрый день, 
Большое спасибо, подсказки очень полезные, у меня получилось немного подебажить и продвинуться, после установки реальных параметров баз, мигать стал не раз в секунду, а аж два раза в секунду :)).
Но почему то пока что координат не получается вытащить.
ANG0    54069   200                             
OBJ0    54069   500
ANG0    54102   200                             
OBJ0    54102   500
ANG0    54135   200                             
OBJ0    54135   500
ANG0    54169   200                             
OBJ0    54169   500
ANG0    54202   200                             
OBJ0    54202   500
ANG0    54235   200                             
OBJ0    54235   500

Хотя данные вроде как уже выглядят лучше
debug> 
Pulse0: 
sensor 0, time 368491501us, len 80 | 
sensor 0, time 368491903us, len 122 | 
sensor 0, time 368499835us, len 69 | 
sensor 0, time 368500241us, len 132 | 
sensor 0, time 368508167us, len 122 | 
sensor 0, time 368508570us, len 60 | 
sensor 0, time 368516502us, len 111 | 
sensor 0, time 368516907us, len 70 | 
sensor 0, time 368524834us, len 59 | 
sensor 0, time 368525237us, len 122 | 
sensor 0, time 368533168us, len 90 | 
sensor 0, time 368533574us, len 111 | 
sensor 0, time 368541501us, len 100 | 
sensor 0, time 368541903us, len 81 | 
sensor 0, time 368549835us, len 111 | 
sensor 0, time 368550241us, len 70 (176 total)
CyclePhaseClassifier: fix 16, phase 3, pulse_base_len 59.288986, history 0x99999999, avg error 0.5 us
PulseProcessor: fix 10, cycle id 4352, num pulses 1 1 0 0, time from last pulse 7950

CyclePhaseClassifier: fix 16, phase 0, pulse_base_len 57.686787, history 0xcccccccc, avg error 1.1 us
PulseProcessor: fix 10, cycle id 8930, num pulses 1 1 0 0, time from last pulse 7928

А комманды 
pp bits
pp angles 
почему то не отрабатывают, хотя в коде они есть.

Не подскажите в чем может быть проблема ?
Плохой входной сигнал ?
Fix 16 пологаю это уже успешный лок ?

ashtuchkin, 2020-09-12:
ANG0 54069 200  <-- kCycleSynced = 200 - значит синхронизационные импульсы мы видим, а позиционные - нет.
OBJ0 54069 500
См вот здесь значения для FixLevel: https://github.com/ashtuchkin/vive-diy-position-sensor/blob/master/include/messages.h#L26
Нужно добиться kFullFix = 1000.

Вот здесь видно что импульсы все длинные (len = 50-150 мкс): 
sensor 0, time 368491501us, len 80 |
sensor 0, time 368491903us, len 122 |
sensor 0, time 368499835us, len 69 |
sensor 0, time 368500241us, len 132 |
В работающей системе должно быть два длинных - один короткий (len = 1-7 мкс), два длинных - один короткий.

Возможно threshold нужно сделать поменьше или станции поближе подвинуть. Удачи!

tech0x, 2020-09-12:
Спасибо, весьма полезные замечания, благодаря этим замечаниям, мне удалось найти положение в котором появились короткие импульсы. В начале у одной станции, а после уменьшения дистанции между станциями в два раза получилось даже получить какую то позицию. Она походу уже валидная ?

ANG0    822502  200     -0.0083 -0.3788 0.0626  -0.4191
OBJ0    822502  1000    0.7580  -1.0367 -0.6598 0.7616
ANG0    822536  200     -0.0023 -0.3764 0.0625  -0.4192
OBJ0    822536  1000    0.7549  -1.0264 -0.6760 0.7424
ANG0    822569  200     0.0053  -0.3751 0.0625  -0.4191
OBJ0    822569  1000    0.7484  -1.0190 -0.6924 0.7154
ANG0    822602  200     0.0139  -0.3725 0.0627  -0.4191
OBJ0    822602  1000    0.7430  -1.0065 -0.7135 0.6851
ANG0    822636  200     0.0227  -0.3712 0.0626  -0.4190
OBJ0    822636  1000    0.7347  -0.9981 -0.7322 0.6537
ANG0    822669  200     0.0308  -0.3717 0.0625  -0.4191
OBJ0    822669  1000    0.7250  -0.9965 -0.7466 0.6231
ANG0    822702  200     0.0383  -0.3714 0.0623  -0.4191
OBJ0    822702  1000    0.7168  -0.9916 -0.7614 0.5953
ANG0    822736  200     0.0461  -0.3672 0.0625  -0.4190
OBJ0    822736  1000    0.7136  -0.9735 -0.7833 0.5702
ANG0    822769  200     0.0539  -0.3605 0.0625  -0.4190
OBJ0    822769  1000    0.7139  -0.9469 -0.8103 0.5485
ANG0    822802  200     0.0620  -0.3516 0.0626  -0.4191
OBJ0    822802  1000    0.7172  -0.9126 -0.8422 0.5279

Вопрос кроме трешолда что еще можно сделать что бы возможно было увеличить дистанцию ?
Попробовать сменить светодиоды ?
Поковырять фильтры помех ?
Один момент который может быть важен, тестирую все это дело ночью, поскольку днем сигнал явно забивается, солнечным светом. Подозреваю что фотоэллементы с алиекспресса, работают по несколько альтернативным принципам.

tech0x, 2020-09-14:
Я нацепил гирлянду из светодиодов в (9 штук) и получилось таки расширить пятно в котором датчик получает данные с баз, но дистанция от базы варирует метр полтора, поэтому можно получить либо с одной базы либо с другой. А вот совместить их вместе пока не получилось. 
Но мне кажется я нашел в чем сложность.
У меня нет чувствительного осцилографа и я подключил ардуинку к выходу с усилителя и поднял счетчик длительности импульсов. И вполне стал регистрировать короткие импульсы длительностью в 16 микросекунд с очевидными интервалами, в тоже время  sensor0 impulses show эти короткие импульсы не регистрировал.
У меня складывается впечатление что здесь лазерный луч имеет также свойства рассеиваться в зависимости от дистанции из-за чего импульсы увеличиваются по длительности из-за чего софт их фильтрует.
Стало бы вопрос, как граммотно поднять верхний лимит на продолжительность коротких импульсов ?
Премного благодарен за Ваш труд.

ashtuchkin, 2020-09-14:
Логика классификации импульсов на короткие и длинные вот здесь https://github.com/ashtuchkin/vive-diy-position-sensor/blob/master/src/pulse_processor.cpp#L44   Все импульсы до 40мкс - короткие, от 40 до 300 - длинные. Но если импульсов нет в "sensor0 impulses show", то проблема раньше, надо смотреть какие входы используются и как. У Teensy есть два варианта входов - компаратор (с threshold) или обычный таймер, можно попробовать переключить на другой вариант и попробовать.

Но в целом я думаю проблема в другом. Мне кажется нужно смотреть схематику фотодиода/усилителя - 16 мкс это слишком много для лазерного луча, не может он так рассеиваться. А вот фотодиод или усилитель может иметь нежелательный RC контур (сопротивление/ёмкость), такой, что сигнал "сглаживается" и вместо четкого короткого импульса получается длинный и гораздо слабее. На осциллографе это было бы хорошо видно.

Я сам когда эту схему делал много экспериментировал с сопротивлениями и конденсаторами, чтобы получить хорошую чувствительность. Думаю с китайскими компонентами это нужно заново делать.

Если не хочется этим заниматься, есть уже настроенные модули с гораздо большей чувствительностью - [TS3633-CM1](https://triadsemi.com/product/ts3633-cm1-prototyping-module/), если есть возможность их заказать то рекомендую.

tech0x, 2020-09-14:
https://github.com/nairol/LighthouseRedox/blob/master/docs/Light%20Emissions.md
Тут указан что короткий синхроимпульс j0 может быть до 62.5 микросекунды в зависимости от положения базы.
E.g. sync pulse j0 can be 2501-3000 ticks long depending on the position and orientation relative to the base station.
Это о другом речь или таки нужно увеличивать размер  для коротких пакетов ?

С сигналом я поработал, форма получилась вполне хорошая, почти идеальная на больших импульсах хорошо видно,
хотя да пришлось перебирать все кондеры и резисторы и подавать на первую ногу усилителя, а не на вторую как в Вашей схеме, поскольку там действительно получался сигнал хуже. Фотоэллементы конечно попробую заказать, просто с этими не понятно толи они их делают то ли нет. А больше  их в других местах не продается.

А насчет коротких импульсов я их заставил ходить весьма варваским ковырянием кода
constexpr TimeDelta min_long_pulse_len(70, usec);

constexpr TimeDelta short_pulse_min_time = angle_center_len - cycle_period / 3;
constexpr TimeDelta short_pulse_max_time = angle_center_len + cycle_period / 3;
Переборирая эти параматры короткие импульсы таки появились, в пульсах. 

Только пока не разобрался, если синхроимпульс может быть таким большим как его сюда правильно стоило бы оформить ?
constexpr TimeDelta short_pulse_min_time = angle_center_len - cycle_period / 3;
constexpr TimeDelta short_pulse_max_time = angle_center_len + cycle_period / 3;

tech0x, 2020-09-14:
Да, так и есть с таким набором параметров, вероятно конечно он не правильный, но мне удалось впервые получить правдоподобные координаты, при условии что базовые станции находятся примерно на 4-4.5 м друг от друга , правда пока только в определенных местах, почему то лучше работает по краям базовых станций. А центра почему то не охотно.  

constexpr TimeDelta min_long_pulse_len(70, usec);
constexpr TimeDelta max_long_pulse_len(280, usec);

constexpr TimeDelta long_pulse_starts_accepted_range(72, usec);
constexpr TimeDelta long_pulse_starts[num_base_stations] = {TimeDelta(0, usec), TimeDelta(410, usec)};

constexpr TimeDelta cycle_period(8333, usec);  // Total len of 1 cycle.
constexpr TimeDelta angle_center_len(4000, usec);
constexpr TimeDelta short_pulse_min_time = angle_center_len - cycle_period / 2;
constexpr TimeDelta short_pulse_max_time = angle_center_len + cycle_period / 2;

OBJ0    664579  1000    0.7672  0.0337  -1.7219 3.7041
OBJ0    664612  1000    0.7671  0.0310  -1.7241 3.7035
OBJ0    664646  800     0.7676  0.0302  -1.7248 3.7031
OBJ0    664679  800     0.7672  0.0294  -1.7258 3.7026
OBJ0    664712  800     0.7666  0.0356  -1.7212 3.7035
OBJ0    664746  800     0.7666  0.0359  -1.7205 3.7042
OBJ0    664779  1000    0.7669  0.0337  -1.7224 3.7034
OBJ0    664812  800     0.7667  0.0324  -1.7231 3.7036
OBJ0    664846  800     0.7672  0.0312  -1.7246 3.7025
OBJ0    664879  800     0.7665  0.0362  -1.7205 3.7037
OBJ0    664912  800     0.7666  0.0352  -1.7213 3.7035
OBJ0    664946  500

OBJ0    664979  500
debug> sensor0 pulses show
Pulse0: accumulating..
debug>.
Pulse0:.
sensor 0, time 671924408us, len 33 |.
sensor 0, time 671928626us, len 238 |.
sensor 0, time 671929052us, len 132 |.
sensor 0, time 671931326us, len 43 |.
sensor 0, time 671936960us, len 283 |.
sensor 0, time 671937384us, len 59 |.
sensor 0, time 671945293us, len 269 |.
sensor 0, time 671945719us, len 69 |.
sensor 0, time 671953627us, len 221 |.
sensor 0, time 671954052us, len 121 |.
sensor 0, time 671957741us, len 33 |.
sensor 0, time 671961960us, len 203 |.
sensor 0, time 671962386us, len 110 |.
sensor 0, time 671964659us, len 43 |.
sensor 0, time 671970293us, len 254 |.
sensor 0, time 671970718us, len 58 (212 total)

ashtuchkin, 2020-09-15:
> Тут указан что короткий синхроимпульс j0 может быть до 62.5 микросекунды в зависимости от положения базы. E.g. sync pulse j0 can be 2501-3000 ticks long depending on the position and orientation relative to the base station. Это о другом речь или таки нужно увеличивать размер для коротких пакетов ?

О другом - там 3 импульса должны регистрироваться каждый цикл (8333 мкс):
 1. Синхроимпульс A - излучается светодиодами базовой станции А, длина ~62.5-300 мкс, начинается в t+0
 2. Синхроимпульс B - излучается светодиодами базовой станции B, длина ~62.5-300 мкс, начинается в t+420мкс
 3. "Позиционный" импульс - излучается лазером одной из станций, длина обычно 2-3 мкс, по положению этого импульса мы определяем угол.

В коде я называю первые 2 импульса "long pulses", третий - "short". Не уверен что изменения в коде помогли - мне кажется они скорее делают так что второй импульс распознается как третий.

Импульсы которые указаны выше мы можем классифицировать вот так:
```
sensor 0, time 671924408us, len 33 |.  t0+4115: импульс 3 (предыдущего цикла)
sensor 0, time 671928626us, len 238 |.  начало цикла t0: импульс 1
sensor 0, time 671929052us, len 132 |.  t0+420: импульс 2
sensor 0, time 671931326us, len 43 |.   t0+2700: импульс 3.
sensor 0, time 671936960us, len 283 |.  начало цикла t0: импульс 1
sensor 0, time 671937384us, len 59 |.   t0+420: импульс 2
sensor 0, time 671945293us, len 269 |.  начало цикла t0: импульс 1
sensor 0, time 671945719us, len 69 |.   t0+420: импульс 2
sensor 0, time 671953627us, len 221 |.  начало цикла t0: импульс 1
sensor 0, time 671954052us, len 121 |.  t0+420: импульс 2
sensor 0, time 671957741us, len 33 |.   t0+4114: импульс 3
sensor 0, time 671961960us, len 203 |.  начало цикла t0: импульс 1
sensor 0, time 671962386us, len 110 |.  t0+420: импульс 2
sensor 0, time 671964659us, len 43 |.   t0+2699: импульс 3
sensor 0, time 671970293us, len 254 |.  начало цикла t0: импульс 1
sensor 0, time 671970718us, len 58      t0+420: импульс 2
```
Что мы видим здесь:
 Импульсы 1 и 2 имеют длину 58 - 283 мкс
 Импульс 3 регистрируется только от одной базовой станции и длина его ~33-43 мкс. Скорее всего усилители "размазывают" сигнал, либо базовая станция слишком близко находится. Большая длина здесь это плохо, т.к. это напрямую отражается на точности измерения угла => точности измерения позиции.

По поводу изменений в коде:
constexpr TimeDelta min_long_pulse_len(70, usec);      <- это threshold между синхро и позиционными импульсами. Исходя из примера выше, лучше поставить 50.
constexpr TimeDelta max_long_pulse_len(280, usec);  <- этот можно увеличить немного т.к. некоторые синхро импульсы шире. Например 300 или 350.

constexpr TimeDelta long_pulse_starts_accepted_range(72, usec);
constexpr TimeDelta long_pulse_starts[num_base_stations] = {TimeDelta(0, usec), TimeDelta(410, usec)};

constexpr TimeDelta cycle_period(8333, usec); // Total len of 1 cycle.
constexpr TimeDelta angle_center_len(4000, usec);
constexpr TimeDelta short_pulse_min_time = angle_center_len - cycle_period / 2;  <- здесь лучше вернуть /3, т.к. иначе короткие и длинные импульсы будут смешиваться. 
constexpr TimeDelta short_pulse_max_time = angle_center_len + cycle_period / 2;  <- тоже лучше вернуть.

Почему /3? Внутри базовой станции лазеры вращаются синхронизировано с циклом 8333 мкс. За один цикл они делают 180 градусов поворот, причем так, чтобы в t0+4000 лазер светил прямо по центру. Т.к. светит станция на 120 градусов вперед (-60 до +60), то мы рассчитываем минимальную и максимальную позицию импульса в цикле как (4000 - 8333 * 60 / 180) и 4000 + (8333 * 60 / 180). Из этого и происходит cycle_period / 3. Побочный эффект этой проверки в том, что синхроимпульсы не конфликтуют с позиционным импульсом: short_pulse_min_time = 1222 мкс, больше чем 420 + max_long_pulse_len.

Далее, вот это кажется как будто фикс, но скорее всего нет, т.к. последнее значение слишком большое. Это расстояние между рассчитанными линиями от базовых станций, в метрах. Это значение примерно показывает точность определения координат и должно быть ~0.01.
OBJ0 664579 1000 0.7672 0.0337 -1.7219 3.7041

tech0x, 2020-09-15:
Да, похоже Вы абсолютно правы, 
я таки нашел короткий импульс осцилографом похоже уровень сигнала становиться недостаточным для усиления.
Благодарю, за уделенное время.

closed 2020-09-15 by tech0x

ashtuchkin, 2020-09-16:
Рад был помочь, удачи!