ashtuchkin/vive-diy-position-sensor #74: Increase precision

toncho11, 2020-09-29:
Hi,

How can the precision be increased? It seems like 1 cm which is not enough for us.

ashtuchkin, 2020-09-29:
It can't be increased, you'll have to use a different solution.

toncho11, 2020-09-30:
But you say:

"3d position accuracy: currently ~10mm; less than 2mm possible with additional work."

What is this additional work?

LeeCookGHA, 2020-09-30:
Increase the number of photo detectors, some additional sensors and then use
https://github.com/cntools/libsurvive for pose estimation.

ashtuchkin, 2020-09-30:
Yep you'll need multiple sensors, certainly not DIY (TS3633-CM1
<https://www.triadsemi.com/product/ts3633-cm1/> work much better), and then
you'll need to do additional research on how to fuse their data.

closed 2021-07-06 by ashtuchkin

luoxz-ai, 2022-01-15:
Can't buy TS3633-CM1 devices anymore