ashtuchkin/vive-diy-position-sensor #75: Guide for a complete newbie

IMKIERA, 2020-12-15:
Hello!

As I said, I'm a complete newbie to the whole programming world but I'm so desperate in working with this project. As of starting I read the Readme file and I started with understanding the software part. Viz; CMake, GNU tool, Ninja and also Eclipse. But according to me, I haven't progressed a bit, how much ever I study or see tutorials on youtube I'm unable to go through this project but somehow I managed to understand what and how each of these software works. 

And I'm working with **STM32 board** and I don't even know how to get that whole programme transferred to this or write everything. 

So all in all, need a complete starter guide to get started with this project for posterity. 
Because, I think there might be many like me out there in hunt of this info! 
I know it'll be hard to give a complete starter guide but at least some guide links would be of much help. 

Thanks in advance 
To all whoever trynna put effort to answer this.

greyham1, 2021-02-25:
I started to build a [EMG ](https://web.archive.org/web/20100621033130/http://www.open-ecg-project.org/tiki-read_article.php?articleId=1) awhile ago. You should learn to solder. it will help.
this project uses a teensey, and a STM32 is not that afaik. If you use any other board, you will need to learn to code
1 Buy the materials from digikey or somewhere, then use the pictures to assemble them. Buy 2 sets 
2 get [arduino IDE](https://www.arduino.cc/en/main/OldSoftwareReleases)
3 use the code from this project to program it using a [FTTI](https://www.aliexpress.com/item/10000348302604.html)
4 ???
5 connect it to steam somehow

GRJ1, 2021-02-25:
I started to build a [EMG ](https://web.archive.org/web/20100621033130/http://www.open-ecg-project.org/tiki-read_article.php?articleId=1) awhile ago. You should learn to solder. it will help.
this project uses a teensey, and a STM32 is not that afaik. If you use any other board, you will need to learn to code
1 Buy the materials from digikey or somewhere, then use the pictures to assemble them. Buy 2 sets 
2 get [arduino IDE](https://www.arduino.cc/en/main/OldSoftwareReleases)
3 use the code from this project to program it using a [FTTI](https://www.aliexpress.com/item/10000348302604.html)
4 ???
5 connect it to steam somehow

IMKIERA, 2021-02-25:
Thank you so much @GRJ1 

Btw, After all these days I've managed to get quite ahead with the project and now stuck with understanding coding part! Theory part I've an idea. but to write program I feel like I'm playing with merry-go-round. (haha) As I said I'm new to coding, I'm not even getting a single meaning out of program. I mean I'm not able to understand the program part even though how much ever I study with Intro to C, C#...  

basically I need info, On explaining program into some understandable English. At least kinda flow chart might also be helpful. If somehow I get through I'll try to contribute here.
  
If anyone have any suggestions open to it.

cheers!

GRJ1, 2021-02-25:
If you follow the code while it's running issuing eclipse or your chosen ide you should be able to see the order the code is executing and which functions can which functions and do what. 
once you find the initial file it should be just a matter of time as long as you have all the libraries installed
I doubt anyone's going to provide a map but there are other people who've used this code if you google

IMKIERA, 2021-02-26:
Thanks again @GRJ1 
That's really helpful.

closed 2021-07-06 by ashtuchkin