ashtuchkin/vive-diy-position-sensor #80: "'reinterpret_cast' from integer to pointer" error during compiling

XDAChen, 2022-02-03:
Hi,
I was getting this error message during compiling:

> error: 'reinterpret_cast' from integer to pointer
   53 |     {&FTM0_SC, IRQ_FTM0, {{/*25*/NA, /*44*/22}, {/*26*/33, /*45*/23},  {/*27*/24, /*46*/ 9}, {      NA, /*49*/10},
make[2]: *** [platform-teensy/CMakeFiles/vive-diy-position-sensor.dir/input_ftm.cpp.obj] Error 1

I then went to the code and changed from:

`// FTM definitions: Ports, Interrupts and Input pins.
constexpr static FTMDef timer_defs[] = {
#if defined(__MK20DX128__) || defined(__MK20DX256__)  // Teensy 3.0, 3.1, 3.2. Chip 64 LQFP pin numbers in comments. Teensy digital pins as numbers.
    {&FTM0_SC, IRQ_FTM0, {{/*25*/NA, /*44*/22}, {/*26*/33, /*45*/23},  {/*27*/24, /*46*/ 9}, {      NA, /*49*/10},`

to the following:

`// FTM definitions: Ports, Interrupts and Input pins.
static FTMDef timer_defs[] = {
#if defined(__MK20DX128__) || defined(__MK20DX256__)  // Teensy 3.0, 3.1, 3.2. Chip 64 LQFP pin numbers in comments. Teensy digital pins as numbers.
    {&FTM0_SC, IRQ_FTM0, {{/*25*/NA, /*44*/22}, {/*26*/33, /*45*/23},  {/*27*/24, /*46*/ 9}, {      NA, /*49*/10},`

After I deleted _constexpr_ it did successfully compiled and I was able to upload the hex file to my teensy3.2 via Tycommander GUI.

The serial port currently outputs nothing, I am going to build the light sensor and set up vive lighthouse and see if I can get data.

I had checked online it seems that the later version of GCC gives this as an error rather than a warning. I am very rudimentary to cpp so I wonder if there is a better solution than a) changing the code like I did (not sure if it works) or b) installing an older version of GCC?

ashtuchkin, 2022-02-03:
Sorry it's been a while since I last compiled this project and I don't have ARM GCC installed now, so hard to advise anything. Hope what you did works.

XDAChen, 2022-02-03:
Hi Alexander, wondering if you can give some clarification on what the line is declaring:

`// FTM definitions: Ports, Interrupts and Input pins. static FTMDef timer_defs[] = { #if defined(__MK20DX128__) || defined(__MK20DX256__) // Teensy 3.0, 3.1, 3.2. Chip 64 LQFP pin numbers in comments. Teensy digital pins as numbers. {&FTM0_SC, IRQ_FTM0, {{/*25*/NA, /*44*/22}, {/*26*/33, /*45*/23}, {/*27*/24, /*46*/ 9}, { NA, /*49*/10},`

So far I am not getting any serial data in at my monitor. I wonder when I deleted constexpr it did not successor define the pins on the teensy-board. 

Thanks!

ashtuchkin, 2022-02-04:
Those are hardware timer module definitions for different versions of teensy. For each timer, it contains interrupt number, pin numbers, etc.

XDAChen, 2022-02-04:
thanks Alex. I got successfully compiled and upload the original .Hex file onto the my teensy board. The board flashes its LED twice rapidly and periodically. However still not receiving any serial data on my Tycommander screen (I had tried Arduino + teensyduino, nothing shows up neither).
Any clue to this problem?

Thanks

ashtuchkin, 2022-02-06:
Are you referring to this manual? https://github.com/ashtuchkin/vive-diy-position-sensor/blob/docs/docs/manual.md  I've started to write it back in the day, but didn't have time to finish. I don't have anything else, sorry. There's some more info in the issues like #65, #73.

XDAChen, 2022-02-06:
I could pull in a more elaborate explanation for it :-)

ashtuchkin, 2022-02-17:
Sorry it's been a long time since I last touched this project, so it would take some time for me to reestablish the context, which I don't have. Let me know if you'd like to contribute a more elaborate explanation so that other people can use it.

XDAChen, 2022-02-21:
Hi Alex,

Yes I would love to provide a more complete description for others. For now
I am working on other parts of my project and will circle back to it very
soon.

Thanks