ashtuchkin/vive-diy-position-sensor #82: How to use tycmd in windows ?

luoxz-ai, 2022-02-17:
How to use tycmd in windows ?

How can I configure the module as you stated in https://github.com/ashtuchkin/vive-diy-position-sensor/blob/docs/docs/manual.md ?

Can the arduino serial port be able to view the debug output information?

luoxz-ai, 2022-02-17:
upload  releases tag v1.0-alpha2    vive-diy-position-sensor.hex ,   how to output the position to the screen？

ashtuchkin, 2022-02-17:
> How to use tycmd in windows ?
I don't know, I don't use windows.

> How can I configure the module as you stated in https://github.com/ashtuchkin/vive-diy-position-sensor/blob/docs/docs/manual.md ?
When you establish serial connection, press enter, it should show you the prompt ("> "). After that, enter your hardware configuration as written in the manual. See also #65, #73.

> Can the arduino serial port be able to view the debug output information?
No, only serial over usb at this point.

> upload releases tag v1.0-alpha2 vive-diy-position-sensor.hex , how to output the position to the screen？
After you connect your teensy/particle to the the sensor and enter the hardware configuration via serial-over-usb, it will wait for the "fix" and start outputting position to whatever port you set in the hardware configuration.

luoxz-ai, 2022-02-17:
[vive_sensor_breadboard.pdf](https://github.com/ashtuchkin/vive-diy-position-sensor/files/8086101/vive_sensor_breadboard.pdf)

luoxz-ai, 2022-02-17:
The arduino is also connected through the usb serial port. Can you interact with the command to get the location debugging information and output it to the screen?

luoxz-ai, 2022-02-17:
https://www.pjrc.com/teensy/tutorial3.html

luoxz-ai, 2022-02-17:
https://www.pjrc.com/teensy/usb_serial.html

[serial_install.exe](https://www.pjrc.com/teensy/serial_install.exe)

USB: Virtual Serial Port？

luoxz-ai, 2022-02-17:
https://github.com/Koromix/tytools/releases

#53 

Use [tytools](https://github.com/Koromix/tytools) (if i remember correctly the command should be 'tycmd monitor') with this you acces the teensy, to exit debug mode (in case you start in it) type '!' (this should put you in config mode).
than you can use the above linked commands in said config mode.

ashtuchkin, 2022-02-17:
For schematics, please use the one in the README.md (right below the words "Full schematics:").
This project does not support Arduino. Only Teensy or Particle.

> Can you interact with the command to get the location debugging information and output it to the screen?

Yes. Specifically the following configuration line does it (note "usb_serial"):
```
stream0 position object0 > usb_serial
```

I'm not sure what you're asking in the other questions.

luoxz-ai, 2022-02-21:
[image]