ashtuchkin/vive-diy-position-sensor #88: ESP32 integration

PaulKokhanov1, 2024-12-25:
I had a quick question about all the needed aspects of the code that would need to change in order to have the 3d positioning work with an esp32 micro-controller? I understand the platform code will need to change but is there any further insight into which parts will need to be redone? Has anyone already began working on such a re-implementation?

Thanks.

ashtuchkin, 2024-12-28:
It's been a long time since I looked into it, but yes, platform integration + you'll need to make sure that esp32 has high-quality hardware timers that can measure at least 0.5-1 microsecond resolution.

closed 2024-12-28 by ashtuchkin

PaulKokhanov1, 2024-12-28:
Thanks for the response Alex, I just had a follow up question. I decided to try to re-implement the 3d position tracking on the esp32 via the Arduino IDE and I have been noticing some strange behavior. I'd hope you may be able to give me further insight. 
Basically, here is a sample of the pulses that I am seeing,

Index 0: Width = 137 us, StartTime = 104756328 us
Index 1: Width = 5 us, StartTime = 104758195 us
Index 2: Width = 97 us, StartTime = 104763468 us
Index 3: Width = 85 us, StartTime = 104764662 us
Index 4: Width = 8 us, StartTime = 104766918 us
Index 5: Width = 9 us, StartTime = 104769471 us
Index 6: Width = 65 us, StartTime = 104771804 us
Index 7: Width = 75 us, StartTime = 104772995 us
Index 8: Width = 7 us, StartTime = 104774861 us
Index 9: Width = 9 us, StartTime = 104777348 us
Index 10: Width = 75 us, StartTime = 104780136 us
Index 11: Width = 106 us, StartTime = 104781329 us
Index 12: Width = 9 us, StartTime = 104783587 us
Index 13: Width = 65 us, StartTime = 104788471 us
Index 14: Width = 116 us, StartTime = 104789663 us
Index 15: Width = 5 us, StartTime = 104791528 us
Index 16: Width = 97 us, StartTime = 104796803 us
Index 17: Width = 85 us, StartTime = 104797996 us
Index 18: Width = 8 us, StartTime = 104800257 us
Index 19: Width = 9 us, StartTime = 104802810 us
Index 20: Width = 85 us, StartTime = 104805139 us
Index 21: Width = 96 us, StartTime = 104806330 us
Index 22: Width = 6 us, StartTime = 104808194 us
Index 23: Width = 8 us, StartTime = 104810686 us

As you can see I am nearly getting the expected Long Long Short pulse pattern, however there seems to be extra short pulses sometimes being picked up, would you happen to know why this is the case?
More importantly though is that as you can see the cycle order is not as expected, for example, if we ignore the two consecutive short pulses and just examine the long pulses we see the following pattern:
B sweep, A sweep, A sweep, A sweep, B sweep, etc...
this is obviously not the expected AABB pattern. 

Within this question (https://github.com/ashtuchkin/vive-diy-position-sensor/issues/26) I saw you indicated that the individual may be missing some cycles, now I don't believe this is the case in my situation because none of the start time differences are greater than the cycle length of 8333 usec thus I am not too sure what could be the issue. If you could provide any insight or potential further considerations I'd really appreciate it! Thanks :)

I do know that the ESP32 does have hardware timers that can measure with a 1 usec resolution

ashtuchkin, 2024-12-30:
> As you can see I am nearly getting the expected Long Long Short pulse pattern, however there seems to be extra short pulses sometimes being picked up, would you happen to know why this is the case?

One guess is there might be something acting as a mirror nearby. Another guess is something with electronics - extra capacitance somewhere, etc. It took me some time with oscilloscope to figure out good thresholds/amplification factor.

> B sweep, A sweep, A sweep, A sweep, B sweep, etc...this is obviously not the expected AABB pattern.

Ok I see a good 8333us pattern, but the distance between first and second long pulses is ~1200us, which is much larger than 400us I got. Are you using original HTC Vive base stations? Maybe they changed something in their protocol in the last 7 years? Not sure.

PaulKokhanov1, 2024-12-30:
> One guess is there might be something acting as a mirror nearby. Another guess is something with electronics - extra capacitance somewhere, etc. It took me some time with oscilloscope to figure out good thresholds/amplification factor.

Ya I was suspecting that, could even the reflection off the resistor cause this? I'll have to experiment more.

> Ok I see a good 8333us pattern, but the distance between first and second long pulses is ~1200us, which is much larger than 400us I got. Are you using original HTC Vive base stations? Maybe they changed something in their protocol in the last 7 years? Not sure.

Ya I'm using the original HTC Vive base stations, its possible they could have changed something but I do feel that's unlikely as I haven't heard/seen anything mentioned within other individuals issues within this repo, but again i'll look into this. Another thing I should've mentioned is this what was I saw on an oscilloscope,

[image]

Note this was done with 1 Base Station in the above image. One thing I couldn't find online is the refresh rate of a single BS? Are we still going through an entire cycle in 8.333 ms? Or does it change when dealing with only 1? Furthermore, the long pulse seems to be around 320 usec in this case which seems slightly large? Could this be due to a faulty circuit? I may need to create a pcb for it. Thanks again for your help.